rockchip-linux/gstreamer-rockchip #2: mpph264enc: unusable output

mcerveny, 2017-04-02:
Hello.
I prepared raw hevc stream. The stream plays okay over DRM but transcoding to h264 makes unusable (corrupted) stream on TinkerBoard (rk3288).
```
ffmpeg -i bbb-1920x1080-cfg06.mkv -c:v copy -an -bsf:v hevc_mp4toannexb 2k.hevc
gst-launch-1.0 filesrc location=2k.hevc ! h265parse ! video/x-h265,framerate=30/1 ! mppvideodec ! kmssink connector-id=65 plane-id=49
gst-launch-1.0 filesrc location=2k.hevc ! h265parse ! video/x-h265,framerate=30/1 ! mppvideodec ! mpph264enc ! filesink location=2k.h264 
```
mpp,libdrm-rockchip,gstreamer-rockchip compiled from today state of https://github.com/rockchip-linux .
kernel is from TInkerOS 1.4 (4.4.16 #25). 

Any hints or patches ? 

Thanks, Martin

PS: If the framerate is not set the program crash with "Floating point exception".

hizukiayaka, 2017-04-05:
The max resolution for H.264(AVC) encoder is 1920x1080.

mcerveny, 2017-04-05:
Yes, "2k.hevc" is 1920x1080 (see the first line) and videotestsrc is 1920x1080 too (see pipeline). If the larger resolution is used pipline fails.

hizukiayaka, 2017-04-05:
It works fine for me, I didn't meet any problem with that.
Maybe the gstreamer rockchip shipped with TinkerOS is too old.

mcerveny, 2017-04-05:
Still buggy for me.
Take TinkerBoard, install fresh copy of TinkerOS 1.6, play **pipe "videotestsrc"**, and test output at least with ffprobe:
```
[h264 @ 8099680] top block unavailable for requested intra4x4 mode -1 at 1 0
[h264 @ 8099680] error while decoding MB 1 0, bytestream 32299
[h264 @ 8099680] concealing 8160 DC, 8160 AC, 8160 MV errors in I frame
[h264 @ 8099680] error while decoding MB 107 53, bytestream -30
[h264 @ 8099680] Cannot use next picture in error concealment
[h264 @ 8099680] concealing 1742 DC, 1742 AC, 1742 MV errors in P frame
[h264 @ 8099680] error while decoding MB 102 55, bytestream -43
[h264 @ 8099680] Cannot use next picture in error concealment
[h264 @ 8099680] concealing 1507 DC, 1507 AC, 1507 MV errors in P frame
[h264 @ 8099680] error while decoding MB 94 60, bytestream -57
[h264 @ 8099680] Cannot use next picture in error concealment
[h264 @ 8099680] concealing 915 DC, 915 AC, 915 MV errors in P frame
Input #0, h264, from '2k.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p, 1920x1080, 30 fps, 30 tbr, 1200k tbn, 60 tbc
```

Play **pipe "mppvideodec ! mpph264enc"** and test output at least with ffprobe:
```
[h264 @ 8099680] top block unavailable for requested intra4x4 mode -1 at 0 0
[h264 @ 8099680] error while decoding MB 0 0, bytestream 32694
[h264 @ 8099680] concealing 8160 DC, 8160 AC, 8160 MV errors in I frame
[h264 @ 8099680] error while decoding MB 73 41, bytestream -17
[h264 @ 8099680] Cannot use next picture in error concealment
[h264 @ 8099680] concealing 3216 DC, 3216 AC, 3216 MV errors in P frame
Input #0, h264, from '2k_tos16.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p, 1920x1080, 30 fps, 30 tbr, 1200k tbn, 60 tbc
```

I also **BUILT** again mpp/for_linux,libdrm-rockchip,gstreamer-rockchip from git  https://github.com/rockchip-linux. Reinstall packages:
```
dpkg -i librockchip-mpp1_1.3.1-1_armhf.deb librockchip-mpp-static_1.3.1-1_armhf.deb librockchip-mpp-dev_1.3.1-1_armhf.deb 
dpkg -i libdrm2_2.4.74-2_armhf.deb libdrm-rockchip1_2.4.74-2_armhf.deb libdrm-radeon1_2.4.74-2_armhf.deb libdrm-nouveau2_2.4.74-2_armhf.deb libdrm-exynos1_2.4.74-2_armhf.deb libdrm-freedreno1_2.4.74-2_armhf.deb libdrm-amdgpu1_2.4.74-2_armhf.deb libdrm-dev_2.4.74-2_armhf.deb
dpkg -i gstreamer1.0-rockchip1_1.10-5_armhf.deb 
```
Play **pipe "mppvideodec ! mpph264enc"** and get the **same errors** as for TinkerOS.

hizukiayaka, 2017-04-06:
It works fine for me.
```
[randy@randy-PC ~]$ ffprobe test.avc 
ffprobe version 3.1.7 Copyright (c) 2007-2017 the FFmpeg developers
  built with gcc 6.3.1 (GCC) 20161221 (Red Hat 6.3.1-1)
  configuration: --prefix=/usr --bindir=/usr/bin --datadir=/usr/share/ffmpeg --incdir=/usr/include/ffmpeg --libdir=/usr/lib64 --mandir=/usr/share/man --arch=x86_64 --optflags='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic' --extra-ldflags='-Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld' --enable-bzlib --disable-crystalhd --enable-fontconfig --enable-frei0r --enable-gcrypt --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libcdio --enable-indev=jack --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libmp3lame --enable-nvenc --extra-cflags=-I/usr/include/nvenc --enable-openal --enable-opencl --enable-libopencv --enable-opengl --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-libschroedinger --enable-libsoxr --enable-libspeex --enable-libtheora --enable-libvorbis --enable-libv4l2 --enable-libvpx --enable-libx264 --enable-libx265 --enable-libxvid --enable-x11grab --enable-avfilter --enable-avresample --enable-postproc --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --disable-stripping --shlibdir=/usr/lib64 --enable-libmfx --enable-runtime-cpudetect
  libavutil      55. 28.100 / 55. 28.100
  libavcodec     57. 48.101 / 57. 48.101
  libavformat    57. 41.100 / 57. 41.100
  libavdevice    57.  0.101 / 57.  0.101
  libavfilter     6. 47.100 /  6. 47.100
  libavresample   3.  0.  0 /  3.  0.  0
  libswscale      4.  1.100 /  4.  1.100
  libswresample   2.  1.100 /  2.  1.100
  libpostproc    54.  0.100 / 54.  0.100
Input #0, h264, from 'test.avc':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p, 720x480, 30 fps, 30 tbr, 1200k tbn, 60 tbc

```

mcerveny, 2017-04-06:
Not for me even 720x480 is buggy.
Let's make full version check and fix for TinkerOS1.6. My versions are bellow and yours testing environment ?

"some commit id disclosured in gst" 
mpi: mpp version: d4c754c author: Randy Li [test]: fix the align problem in decode_advanced()
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 0 [0 : 0]
```
# uname -a
Linux linaro-alip 4.4.16 #1 SMP Thu Mar 30 22:22:19 CST 2017 armv7l GNU/Linux

# grep PRETTY_NAME /etc/os-release
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"

# dpkg -l | egrep '(ffmpeg|rockchip|gstreamer)'
ii  ffmpeg                                   7:3.2.4-1~bpo8+1                          armhf        Tools for transcoding, streaming and playing of multimedia files
ii  gir1.2-gstreamer-1.0                     1.10.4-1                                  armhf        GObject introspection data for the GStreamer library
ii  gstreamer1.0-alsa:armhf                  1.4.4-2+deb8u1                            armhf        GStreamer plugin for ALSA
ii  gstreamer1.0-libav:armhf                 1.10.4-1                                  armhf        libav plugin for GStreamer
ii  gstreamer1.0-plugins-bad:armhf           1.10.4-1                                  armhf        GStreamer plugins from the "bad" set
ii  gstreamer1.0-plugins-base:armhf          1.10.4-1                                  armhf        GStreamer plugins from the "base" set
ii  gstreamer1.0-plugins-good:armhf          1.4.4-2+deb8u3                            armhf        GStreamer plugins from the "good" set
ii  gstreamer1.0-pulseaudio:armhf            1.4.4-2+deb8u3                            armhf        GStreamer plugin for PulseAudio
ii  gstreamer1.0-rockchip1                   1.10-5                                    armhf        The Gstreamer plugins for Rockchip RK3xxx platform.
ii  gstreamer1.0-tools                       1.4.4-2+deb8u1                            armhf        Tools for use with GStreamer
ii  gstreamer1.0-x:armhf                     1.4.4-2+deb8u1                            armhf        GStreamer plugins for X11 and Pango
ii  libgstreamer-plugins-bad1.0-0:armhf      1.10.4-1                                  armhf        GStreamer development files for libraries from the "bad" set
ii  libgstreamer-plugins-base1.0-0:armhf     1.10.4-1                                  armhf        GStreamer libraries from the "base" set
ii  libgstreamer1.0-0:armhf                  1.10.4-1                                  armhf        Core GStreamer libraries and elements
ii  librockchip-mpp-dev                      1.3.1-1                                   armhf        Media Process Platform
ii  librockchip-mpp-static                   1.3.1-1                                   armhf        Media Process Platform
ii  librockchip-mpp1                         1.3.1-1                                   armhf        Media Process Platform
ii  librockchip-vpu0                         1.3.1-1                                   armhf        Media Process Platform

# gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12,width=1920,height=1080 ! mpph264enc ! filesink location=1920.h264
Setting pipeline to PAUSED ...
mpi: mpp version: d4c754c author: Randy Li [test]: fix the align problem in decode_advanced()
Pipeline is PREROLLING ...
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 0 [0 : 0]
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:07.493341846
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
mpp_thread: MPP_THREAD_STOPPING status set mThread 0xbdbd60
Freeing pipeline ...

# ffprobe -v debug 1920.h264 
ffprobe version 3.2.4-1~bpo8+1 Copyright (c) 2007-2017 the FFmpeg developers
  built with gcc 4.9.2 (Debian 4.9.2-10)
...
  libavutil      55. 34.101 / 55. 34.101
  libavcodec     57. 64.101 / 57. 64.101
  libavformat    57. 56.101 / 57. 56.101
  libavdevice    57.  1.100 / 57.  1.100
  libavfilter     6. 65.100 /  6. 65.100
  libavresample   3.  1.  0 /  3.  1.  0
  libswscale      4.  2.100 /  4.  2.100
  libswresample   2.  3.100 /  2.  3.100
  libpostproc    54.  1.100 / 54.  1.100
[file @ 0x80f4efa0] Setting default whitelist 'file,crypto'
[h264 @ 0x80f4e930] Format h264 probed with size=2048 and score=51
[h264 @ 0x80f4e930] Before avformat_find_stream_info() pos: 0 bytes read:32768 seeks:0 nb_streams:1
[h264 @ 0x80f4f9d0] nal_unit_type: 7, nal_ref_idc: 1
[h264 @ 0x80f4f9d0] nal_unit_type: 8, nal_ref_idc: 1
[h264 @ 0x80f4f9d0] nal_unit_type: 6, nal_ref_idc: 0
[h264 @ 0x80f4f9d0] nal_unit_type: 5, nal_ref_idc: 1
[h264 @ 0x80f4f9d0] Reinit context to 1920x1088, pix_fmt: yuv420p
[h264 @ 0x80f4f9d0] concealing 8160 DC, 8160 AC, 8160 MV errors in I frame
[h264 @ 0x80f4f9d0] nal_unit_type: 1, nal_ref_idc: 1
[h264 @ 0x80f4f9d0] bytestream overread -49
[h264 @ 0x80f4f9d0] error while decoding MB 107 53, bytestream -49
[h264 @ 0x80f4f9d0] concealing 1742 DC, 1742 AC, 1742 MV errors in P frame
[h264 @ 0x80f4f9d0] nal_unit_type: 1, nal_ref_idc: 1
[h264 @ 0x80f4f9d0] bytestream overread -60
[h264 @ 0x80f4f9d0] error while decoding MB 102 55, bytestream -60
[h264 @ 0x80f4f9d0] concealing 1507 DC, 1507 AC, 1507 MV errors in P frame
[h264 @ 0x80f4f9d0] nal_unit_type: 1, nal_ref_idc: 1
[h264 @ 0x80f4f9d0] bytestream overread -66
[h264 @ 0x80f4f9d0] error while decoding MB 94 60, bytestream -66
[... cut ...]
[h264 @ 0x80f4e930] After avformat_find_stream_info() pos: 3932160 bytes read:3932160 seeks:0 frames:120
Input #0, h264, from '1920.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0, 120, 1/1200000: Video: h264 (High), 1 reference frame, yuv420p(progressive, left), 1920x1080 (1920x1088), 0/1, 30 fps, 30 tbr, 1200k tbn, 60 tbc
[h264 @ 0x80f564c0] nal_unit_type: 7, nal_ref_idc: 1
[h264 @ 0x80f564c0] nal_unit_type: 8, nal_ref_idc: 1
[AVIOContext @ 0x80f57150] Statistics: 3932160 bytes read, 0 seeks
```

hizukiayaka, 2017-04-06:
confirmed

hizukiayaka, 2017-05-17:
It should be fixed in the recent version.

closed 2017-06-01 by hizukiayaka

Climax777, 2018-05-10:
Something similar is happening to me on rock64. The encoder "works" but ffprobe cannot find a working h264 frame.