rockchip-linux/gstreamer-rockchip #3: gstreamer dependecies

mcerveny, 2017-04-06:
Compilation on GST 1.8.3 fails (gstmppvideodec.c:124:14: error: ‘GST_VIDEO_FORMAT_P010_10LE’ undeclared). Dependency is set on GSTPB=1.8.2 (https://github.com/rockchip-linux/gstreamer-rockchip/blob/develop/configure.ac#L50) but symbol GST_VIDEO_FORMAT_P010_10LE exists from 1.9.1 (see commit https://github.com/GStreamer/gst-plugins-base/commit/079ceb894cd73bde243820231f714a5b6a7801f9).

hizukiayaka, 2017-04-06:
Actually it is not P010_10LE, I call it P010_10LEC.

mcerveny, 2017-04-06:
???? I see only GST_VIDEO_FORMAT_P010_10LE https://github.com/rockchip-linux/gstreamer-rockchip/blob/develop/gst/rockchipmpp/gstmppvideodec.c#L124
Is there any updated version of sources ?

hizukiayaka, 2017-04-07:
look at the comment above it

closed 2017-04-10 by hizukiayaka