rockchip-linux/gstreamer-rockchip #5: when make,meet error: " /usr/bin/ld:.libs/libgstkmssink.ver:2: syntax error in VERSION script"

zhang-liye, 2017-05-10:
when make,there are errors, following is make log:

firefly@firefly:~/zhangliye/gstreamer-rockchip$ 
firefly@firefly:~/zhangliye/gstreamer-rockchip$ make
make  all-recursive
make[1]: Entering directory '/home/firefly/zhangliye/gstreamer-rockchip'
Making all in gst-libs
make[2]: Entering directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs'
Making all in gst
make[3]: Entering directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs/gst'
Making all in vpudec
make[4]: Entering directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs/gst/vpudec'
  CCLD     libgstvpu-base.la
ar: `u' modifier ignored since `D' is the default (see `U')
make[4]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs/gst/vpudec'
make[4]: Entering directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs/gst'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs/gst'
make[3]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs/gst'
make[3]: Entering directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs'
make[2]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst-libs'
Making all in gst
make[2]: Entering directory '/home/firefly/zhangliye/gstreamer-rockchip/gst'
Making all in kms
make[3]: Entering directory '/home/firefly/zhangliye/gstreamer-rockchip/gst/kms'
  CC       libgstkmssink_la-gstkmssink.lo
  CC       libgstkmssink_la-gstkmsutils.lo
  CC       libgstkmssink_la-gstkmsallocator.lo
  CC       libgstkmssink_la-gstkmsbufferpool.lo
  CCLD     libgstkmssink.la
/usr/bin/ld:.libs/libgstkmssink.ver:2: syntax error in VERSION script
collect2: error: ld returned 1 exit status
Makefile:596: recipe for target 'libgstkmssink.la' failed
make[3]: *** [libgstkmssink.la] Error 1
make[3]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst/kms'
Makefile:509: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst'
Makefile:560: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip'
Makefile:491: recipe for target 'all' failed
make: *** [all] Error 2
firefly@firefly:~/zhangliye/gstreamer-rockchip$

zhang-liye, 2017-05-10:
who has met such similar problem?  any solution ?

hizukiayaka, 2017-05-11:
--disable-kms

zhang-liye, 2017-05-11:
@hizukiayaka  thank you

hizukiayaka, 2017-05-11:
Does it work for you, then I may close this issue?

zhang-liye, 2017-05-11:
1. if i do this:  ./configure --disable-kms    i still have another similar problem

 CC       libgstrkximage_la-rkx_kmsutils.lo
  CCLD     libgstrkximage.la
/usr/bin/ld:.libs/libgstrkximage.ver:2: syntax error in VERSION script
collect2: error: ld returned 1 exit status
Makefile:587: recipe for target 'libgstrkximage.la' failed
make[3]: *** [libgstrkximage.la] Error 1
make[3]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst/rkximage'
Makefile:509: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip/gst'
Makefile:560: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/firefly/zhangliye/gstreamer-rockchip'
Makefile:491: recipe for target 'all' failed
make: *** [all] Error 2
firefly@firefly:~/zhangliye/gstreamer-rockchip$

2. i try to solve, ./configure --disable-drmrockchip --disable-kms,  however, i cannot "make" anything

configure: *** Plug-ins that have NOT been ported:

configure: *** Plug-ins with dependencies that will be built:

configure: *** Plug-ins with dependencies that will NOT be built:
	drmrockchip
	kms
	rockchipmpp
	vpudec

hizukiayaka, 2017-05-11:
Oh, dear, please don't ask this kind of simple build problem unless you know what you are doing.
Install the rockchip mpp first. And drmrockchip requests libdrm-rockchip.
Both kms and drmrockchip are the video sink. You won't need them as they are not the decoder.
Only the rockchipmpp and vpudec is the decoder. The vpudec has been dropped.

closed 2017-05-11 by hizukiayaka

geekerlw, 2017-05-11:
look at this line: https://github.com/rockchip-linux/gstreamer-rockchip/blob/develop/configure.ac#L331 , make sure you have rockchip mpp installed, kms and drm the same :)