rockchip-linux/gstreamer-rockchip #15: the gstreamer plugin mppvideodec don't work well.

alanlilyan, 2017-09-15:
I want to try to use rk3288 (with kernel 4.4 and Ubuntu 16.04) to be the thin client to decode H.264 stream. My pipeline is 
`appsrc !  h264parse ! mppvideodec ! videoconvert ! appsink`.
It could work, however, the fps maybe only 5 or lower. I find the `/dev/vpu-service`  and `librockchip_vpu.so librockchip_mpp.so libdrm_rockchip.so`. I can see the mpp log info too. 
How can I know whether used the vpu model acceleration? or how to fix this?

wzyy2, 2017-09-15:
I saw `videoconvert`, so CPU might be involved to convert video frames.
You should check your application and reduce CPU workload.
you can refer to this demo:
https://github.com/wzyy2/gstreamer-opencv/blob/master/PipeLines.h

When developing media/graphics application on ARM, Please note that **memory-copying is expensive**.
*  ARM cores are slower than X86 in memory access.
* Video/Graphics buffers usually are uncached DMA-Buffer and uncached reads/writes from CPU are really slow.

alanlilyan, 2017-09-16:
I know memory-copying is not a good way, but  I try to just replace the mppvideodec to avdec_h264, it worked better than mppvideodec (although not better enough). So I think it maybe not the videoconvert caused the problem, but I have seen the demo you mentioned and  I will try it.

wzyy2, 2017-09-16:
"I know memory-copying is not a good way, but I try to just replace the mppvideodec to avdec_h264, it worked better than mppvideodec"

See `Video/Graphics buffers usually are uncached DMA-Buffers and uncached reads/writes from CPU are really slow.`

alanlilyan, 2017-09-17:
Sorry, I know what you mean.  I would like try the demo you given. But on the other hand, I just used avdec_h264 to replace the mppvideodec, and no more other things changed. The most important thing is avdec_h264 as decoder using CPU, and also with the uncached DMA-Buffers.
Maybe something wrong when I installed the Ubuntu OS or compiled the libraries.
Anyway, thanks for your helping.

closed 2017-09-20 by hizukiayaka