rockchip-linux/gstreamer-rockchip #22: ** (gst-launch-1.0:1723): CRITICAL **: gst_mpp_allocator_start: assertion 'size != 0' failed

josephthlee, 2018-03-31:
Hi,

I try to use mppvideodec to decode H264 rtsp stream and I have no error until I Ctrl+C to quit.
What's wrong?

gst-launch-1.0  rtspsrc location="rtspt://192.168.18.3:554/user=admin&password=&channel=1&stream=1.sdp" ! decodebin !  appsink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtspt://192.168.18.3:554/user=admin&password=&channel=1&stream=1.sdp
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
mpi: mpp version: 5849089 author: Herman Chen [mpp]: Add temporally patch for blocking issue
hal_h264d_api: hal_h264d_init mpp_buffer_group_get_internal used ion In
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:09.078548280
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
(CTRL+C)
** (gst-launch-1.0:1723): CRITICAL **: gst_mpp_allocator_start: assertion 'size != 0' failed
0:00:09.863576340  1723   0x7fb003db20 ERROR       mppdecbufferpool gstmppdecbufferpool.c:112:gst_mpp_dec_buffer_pool_start:<mppvideodec0:pool:src> we received 0 buffer, we want at least 22
0:00:09.863695339  1723   0x7fb003db20 ERROR             bufferpool gstbufferpool.c:564:gst_buffer_pool_set_active:<mppvideodec0:pool:src> start failed
0:00:09.863744922  1723   0x7fb003db20 ERROR            mppvideodec gstmppvideodec.c:654:gst_mpp_video_dec_handle_frame:<mppvideodec0> Unable to activate the pool
0:00:09.863790714  1723   0x7fb003db20 ERROR            mppvideodec gstmppvideodec.c:684:gst_mpp_video_dec_handle_frame:<mppvideodec0> can't process this frame
Setting pipeline to NULL ...
Freeing pipeline ...

hizukiayaka, 2018-04-04:
Can't allocate buffer from the system. Which platform and system do you use?

hizukiayaka, 2018-04-09:
Quit old kernel, I would suggest you to use a pre-build image unless you are advanced in this area.

closed 2018-04-09 by hizukiayaka

josephthlee, 2018-04-10:
Thanks for your reply.
Can you advice some Linux image that already has all the hw accels installed? Thank you.