rockchip-linux/gstreamer-rockchip #23: how to configure build to find correct gstreamer

nenadlazovic, 2018-04-28:
I am trying to build gstreamer-rockchip on AsusTinkerboard w OS 20180222-tinker-board-linaro-stretch-alip-v2.0.5.img.

I recompiled latest gstreamer 1.15 using cerbero (https://gstreamer.freedesktop.org/documentation/installing/building-from-source-using-cerbero.html#). Build generates dist/ in gstreamer-1.0/cerbero/build/dist/linux_armv7/*
It runs ok (of course without hardware acceleration).

When trying to compile gstreamer-rockchip, build can not find my gststreamer dist.

Questions: how to configure gstreamer-rockchip build to point to gststreamer dist ?

Thanks.

chandong83, 2018-05-01:
Check the LD_LIBRARY_PATH System Environment.

LD_LIBRARY_PATH must be set to Path Gstreamer Libarary.

$ echo $LD_LIBRARY_PATH
and
$ export LD_LIBRARY_PATH=/usr/local/lib
or
$ export LD_LIBRARY_PATH=/usr/lib

nenadlazovic, 2018-05-01:
chandong83,
I gave up on gstreamer cerbero build. I am using steps from your blog at https://blog.naver.com/PostList.nhn?blogId=chandong83

Following dependence are not available: libdirac-dev libschroedinger-dev libslv2-dev. 
Any suggestions ?

thanks

chandong83, 2018-05-02:
Hi,

I share the PDF file. 
I installed it with this.
but, it is in Chinese.
you can read it in Google Translator. I saw that too.

Thank you.
[RK Linux Camera GStreamer PDF](http://blogattach.naver.net/d247ce7d6f3736eac52244724ea8d3ab0d5ca64791/20180502_183_blogfile/chandong83_1525231392704_b3cl10_pdf/RK_linux_camera_gstreamer_%3F%E9%C4%3F%3F_v1.0_20171212.pdf)

closed 2018-05-02 by hizukiayaka

nenadlazovic, 2018-05-02:
thank you. will try to follow document.

chandong83, 2018-05-02:
can i have your email?
i will send it.

nenadlazovic, 2018-05-02:
nlazovic@senorta.com

nationalflag, 2020-03-22:
@chandong83 @nenadlazovic could you send the pdf to me? Thanks! 595421379@qq.com

kcarriello, 2020-07-09:
I also need this PDF, could you please send it to me. klebercarriello@gmail.com

chandong83, 2020-07-09:
> I also need this PDF, could you please send it to me. [klebercarriello@gmail.com](mailto:klebercarriello@gmail.com)

I uploaded it in the link below.
[https://github.com/chandong83/rockchips_docs/blob/master/RK_linux_camera_gstreamer_%E5%BA%94%E7%94%A8%E5%BC%80%E5%8F%91_v1.0_20171212.pdf](url)