rockchip-linux/gstreamer-rockchip #28: mpph264enc not work in 1080P

andy9100, 2018-06-23:
Hello,
I test mpph264enc to encoder 1080P image with following test script
gst-launch-1.0 videotestsrc ! videoscale ! video/x-raw,width=1920,height=1080 ! mpph264enc ! fakesink
error message:
linaro@linaro-alip:~$ gst-launch-1.0 videotestsrc ! videoscale ! video/x-raw,width=1920,height=1080 ! mpph264enc ! fakesink
Setting pipeline to PAUSED ...
mpi: mpp version: 232eb6d author: kenvenis.chen [h263d] : add vpu1 support
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Pipeline is PREROLLING ...
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 7776000 [7290000 : 8262000]
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988
mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988

test environment:
HW: rk3399
OS: Debian GNU/Linux 9 
Gst version: GStreamer Core Library version 1.12.5

closed 2018-06-23 by andy9100

santo4ul, 2019-10-18:
Hi @andy9100 , I'm facing the same issue. Any idea of how to fix this issue? 

Thanks,
San

stylesuxx, 2020-03-03:
I am sorry, but why has this been closed? Facing the same issue.

gusarg81, 2020-08-24:
Same for me, using Armbian Focal (with legacy kernels) on Rock64, I get:

> mpp_log: Assertion prep->ver_stride == (((prep->height)+(8)-1)&~((8)-1)) failed at h264e_vpu_update_hw_cfg:988

When using ffmpeg and gstreamer for hw encoding:

> ffmpeg -i rtsp://admin:XXXX@10.0.1.10:554/onvif1 -c:v rawvideo -f rawvideo - | gst-launch-1.0 fdsrc ! rawvideoparse format=nv12 width=1920 height=1080 framerate=30/1 ! mpph264enc ! filesink location=test1.mp4

Any ideas?