rockchip-linux/gstreamer-rockchip #43: mpph264enc configuration?

om777, 2019-01-30:
Hi,
Is there any configuration parameters for mpph264enc plugin?
Particularly, how to set CBR mode and choose the target bitrate for it?
I don't see any configuration parameters of plugin mpph264enc.

# gst-inspect-1.0 mpph264enc
Factory Details:
  Rank                     primary + 1 (257)
  Long-name                Rockchip Mpp H264 Encoder
  Klass                    Codec/Encoder/Video
  Description              Encode video streams via Rockchip Mpp
  Author                   Randy Li <randy.li@rock-chips.com>

Plugin Details:
  Name                     rockchipmpp
  Description              Rockchip Mpp Video Plugin
  Filename                 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstrockchipmpp.so
  Version                  1.10.0
  License                  GPL
  Source module            gstreamer-rockchip
  Binary package           GStreamer Plug-ins source release
  Origin URL               Unknown package origin

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstVideoEncoder
                         +----GstMppVideoEnc
                               +----GstMppH264Enc

Implemented Interfaces:
  GstPreset

Pad Templates:
  SINK template: 'sink'
    Availability: Always
    Capabilities:
      video/x-raw
                 format: NV12
                  width: [ 32, 1920 ]
                 height: [ 32, 1088 ]
              framerate: [ 0/1, 60/1 ]
      video/x-raw
                 format: I420
                  width: [ 32, 1920 ]
                 height: [ 32, 1088 ]
              framerate: [ 0/1, 60/1 ]
      video/x-raw
                 format: YUY2
                  width: [ 32, 1920 ]
                 height: [ 32, 1088 ]
              framerate: [ 0/1, 60/1 ]
      video/x-raw
                 format: UYVY
                  width: [ 32, 1920 ]
                 height: [ 32, 1088 ]
              framerate: [ 0/1, 60/1 ]

  SRC template: 'src'
    Availability: Always
    Capabilities:
      video/x-h264
                  width: [ 32, 1920 ]
                 height: [ 32, 1080 ]
              framerate: [ 0/1, 60/1 ]
          stream-format: { byte-stream }
              alignment: { au }
                profile: { high }

Element Flags:
  no flags set

Element Implementation:
  Has change_state() function: gst_video_encoder_change_state

Element has no clocking capabilities.
Element has no URI handling capabilities.

Pads:
  SINK: 'sink'
    Pad Template: 'sink'
  SRC: 'src'
    Pad Template: 'src'

Element Properties:
  name                : The name of the object
                        flags: readable, writable
                        String. Default: "mpph264enc0"
  parent              : The parent of the object
                        flags: readable, writable
                        Object of type "GstObject"

Sufaev, 2019-01-31:
Hello.

I have similar question.
Official documentation of Rockchip MPP hardware says that encoder supports up to high profile 4.0 including main and baseline profile http://opensource.rock-chips.com/images/f/fa/MPP_Development_Reference.pdf

But the description of plugin src tells profile: { high }.

Why plugin does not support other profiles?

SovGVD, 2019-04-02:
Hi

The parameters of encoding were hardcoded to High 4.0
High (100) https://github.com/rockchip-linux/gstreamer-rockchip/blob/484e3f34c2123c9050b822e27f68bf7439c57099/gst/rockchipmpp/gstmpph264enc.c#L137
4.0 (40) https://github.com/rockchip-linux/gstreamer-rockchip/blob/484e3f34c2123c9050b822e27f68bf7439c57099/gst/rockchipmpp/gstmpph264enc.c#L138

And there is no option to change it. For my project I need baseline profile and just changing  `codec_cfg.h264.profile` to 66 (baseline http://opensource.rock-chips.com/wiki_Mpp) give me only one frame and than encoder freezing (and yes, I disable other features, that are not suitable for baseline profile, but look it is not enough).

I will try to provide path as soon as I found best solution for that. Or may be someone will help me with it =)

santo4ul, 2019-10-20:
Hi,

I've the same problem.

Have someone found how to adjust the encoder bitrate?

Thanks,
San

billythedummy, 2020-07-20:
Any chance this fork https://github.com/stylesuxx/gstreamer-rockchip/commit/195e483a1a8dd6777b232b53bf0adc48baa365f4 will be merged soon?

JeffyCN, 2020-08-25:
added in https://github.com/rockchip-linux/gstreamer-rockchip/commit/c796bc1e1db8d5fc28d4c456863175ec81126fae

NOTE: needs the newest mpp.

closed 2020-08-25 by JeffyCN