rockchip-linux/gstreamer-rockchip #46: is mpph264enc 30fps?

doovirus, 2019-07-12:
Hi
i'm testing on aio3288j board.
i tested mpph264enc at 21 fps.
how can i make 30 fps??

next line is testing commands.
**First, the input was confirmed to be 60fps using rkcamsrc.**

root@firefly:~# gst-launch-1.0 -v rkcamsrc device=/dev/video1 ! video/x-raw,format=NV12,width=1920,height=1088,framerate=60/1 ! fpsdisplaysink name=fpssink text-overlay=false video-sink=fakesink signal-fps-measurements=true sync=false
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRKCamSrc:rkcamsrc0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink.GstGhostPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
requesting 4 MMAP buffers/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 31, dropped: 0, current: 61.95, average: 61.95
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 61, dropped: 0, current: 59.94, average: 60.95
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 91, dropped: 0, current: 59.86, average: 60.58
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 122, dropped: 0, current: 60.02, average: 60.44
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 152, dropped: 0, current: 59.94, average: 60.34
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 182, dropped: 0, current: 59.85, average: 60.26
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 213, dropped: 0, current: 59.98, average: 60.22
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 243, dropped: 0, current: 59.99, average: 60.19

**Second, adding mpph264enc**

root@firefly:~# gst-launch-1.0 -v rkcamsrc device=/dev/video1 ! video/x-raw,format=NV12,width=1920,height=1088,framerate=60/1 ! mpph264enc ! fpsdisplaysink name=fpssink text-overlay=false video-sink=fakesink signal-fps-measurements=true sync=false
Setting pipeline to PAUSED ...
mpi: mpp version: a3bdfde0 author: Herman Chen [hal_h264e]: Add gop ref implement to vepu2
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRKCamSrc:rkcamsrc0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 15667200 [14688000 : 16646400]
/GstPipeline:pipeline0/GstMppH264Enc:mpph264enc0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
requesting 2 MMAP buffers/GstPipeline:pipeline0/GstMppH264Enc:mpph264enc0.GstPad:src: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1920, height=(int)1088, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1920, height=(int)1088, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0.GstPad:sink: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1920, height=(int)1088, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink.GstGhostPad:sink: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, width=(int)1920, height=(int)1088, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)60/1, interlace-mode=(string)progressive, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 12, dropped: 0, current: 22.73, average: 22.73
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 24, dropped: 0, current: 21.19, average: 21.93
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 34, dropped: 0, current: 19.44, average: 21.14
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 46, dropped: 0, current: 23.33, average: 21.67
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 57, dropped: 0, current: 21.01, average: 21.54

**Last, io-mode of rkcamsrc option changed to dmabuf.**

root@firefly:~# gst-launch-1.0 -v rkcamsrc device=/dev/video1 io-mode=dmabuf ! video/x-raw,format=NV12,width=1920,height=1088,framerate=60/1 ! mpph264enc ! fpsdisplaysink name=fpssink text-overlay=false video-sink=fakesink signal-fps-measurements=true sync=false
Setting pipeline to PAUSED ...
mpi: mpp version: a3bdfde0 author: Herman Chen [hal_h264e]: Add gop ref implement to vepu2
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink/GstFakeSink:fakesink0: sync = false
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRKCamSrc:rkcamsrc0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1088, framerate=(fraction)60/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)1:4:5:4
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 15667200 [14688000 : 16646400]
[... cut ...]
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 48, dropped: 0, current: 23.05, average: 22.89
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 60, dropped: 0, current: 23.15, average: 22.94
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 71, dropped: 0, current: 20.78, average: 22.57
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 83, dropped: 0, current: 22.71, average: 22.59
/GstPipeline:pipeline0/GstFPSDisplaySink:fpssink: last-message = rendered: 94, dropped: 0, current: 21.70, average: 22.49

jpiat, 2020-01-23:
Hi,
same issue here. I have a camera that outputs 30fps (confirmed though gstreamer), but mpph264enc drops half of the frames resulting in 15fps output.

any idea on how to achieve 30fps (as advertised by Rockchip)?

doovirus, 2020-01-29:
https://github.com/rockchip-linux/mpp/issues/104

Since then my project has been dropped.
Sorry. I have no idea.

jpiat, 2020-02-11:
I managed to get an increase of FPS with using queue on the input and output of the mpph264enc. Still not 30FPS though.

jacksonvm, 2020-07-06:
@jpiat Can you please share the command you are running and what is your final result? How much FPS are you getting?

jpiat, 2020-07-21:
i can do 25FPS but only when limiting the cores frequency to 600MHz ...

JeffyCN, 2020-08-12:
that should due to frame copy + encode time.
using async tasks and bump vcodec aclk freq should help, or implement dma buffer pool in enc plugin to support io-mode dmabuf-import

jpiat, 2020-08-12:
Hi, I made a gstreamer-less version of the encoding using pure v4l2 and MPP
library and the problem remains. I think that there is some sort of bus
contention when using both the ISP and the encoder because if I encode
random buffers at 1080p I get a stable 40 FPS, and if I read the camera
alone in v4l2 I get 30FPS. Using both (reading frame using v4l2 and
encoding using MPP) in a pipeline i get between 25FPS and 12FPS (unstable).
I think that rockchip is aware of the issue because the MPP library as
unpopulated code to do encoding straight from the ISP.

JeffyCN, 2020-08-12:
normally there should be 3 parts:
1/ v4l2 sourcing --- 30fps
2/ frame copy(mostly memcpy)  --- about 40fps for 1080p on 3399
3/ encoding --- about 40~50 fps with 600M freq on 3399

you need to make them all async to reach 30fps, or even skip the 2/ step by importing mpp's dmabuf to v4l2.

jpiat, 2020-08-12:
I used the async interface of the MPP and v4l2 was used in async way also.
I also pass MPP dmabuff straight to v4l2 when getting a frame instead of
having an intermediate copy.

What is very surprising is that when doing the encoding the MPP calls
(pushing a frame and getting a buffer) only take ~25ms, but  the v4l2 part
(grabbing one image from the buffer queue) that's taking between 10ms and
80ms.

My pipeline is the following :

Grab a v4l2 frame
Loop:
   Push frame to MPP
   Grab v4l2 frame
   Pop buffer from MPP

JeffyCN, 2020-08-12:
maybe try perf tool to measure the cpu usage

jpiat, 2020-08-12:
I did, CPU usage is fairly low in top ~50% of one core. If someone can
share a code or gstreamer pipeline that succeeds to get 30FPS@1080p H264
encoding for a camera connected to the ISP delivering a YUYV format I'd be
interested.

JeffyCN, 2021-05-13:
Well, i finally got time to rework this plugin(both of the enc and dec).

So far so good, by reusing the source dma buf directly(needs mplane plugin from v4l-utils for single plane) and doing async encoding, it can reach 1080P@30FPS with YUY2+H264.

LD_PRELOAD=/usr/lib/v4l1compat.so GST_DEBUG=fpsdisplaysink:5 gst-launch-1.0 v4l2src ! mpph264enc ! h264parse ! mppvideodec ! fpsdisplaysink video-sink=waylandsink

0:00:23.388226215  2625   0x7f98019460 DEBUG         fpsdisplaysink fpsdisplaysink
.c:372:display_current_fps:<fpsdisplaysink0> Updated max-fps to 30.195134
0:00:29.216685598  2625   0x7f98019460 DEBUG         fpsdisplaysink fpsdisplaysink
.c:376:display_current_fps:<fpsdisplaysink0> Updated min-fps to 24.216773
0:00:45.485233682  2625   0x7f98019460 DEBUG         fpsdisplaysink fpsdisplaysink
.c:372:display_current_fps:<fpsdisplaysink0> Updated max-fps to 30.327509
0:02:00.795751220  2625   0x7f98019460 DEBUG         fpsdisplaysink fpsdisplaysink
.c:372:display_current_fps:<fpsdisplaysink0> Updated max-fps to 30.345868

It would take a few days to cleanup the patches.

closed 2021-05-17 by JeffyCN