rockchip-linux/gstreamer-rockchip #47: cannot initialize work tree for gstreamer-rockchip

avp24, 2019-07-14:
I am trying to install rockchip linux sdk by following steps mentioned in below link:
http://opensource.rock-chips.com/wiki_Source

Now after giving below command:
../repo/repo sync -f -j10
I am facing below error:

yash4ever, 2019-07-15:
I am trying to buildroot image in my ubuntu 16.04 for  RK3328  i am following

http://opensource.rock-chips.com/wiki_Linux_user_guide

and encountering sync error 

 * [new branch]      master     -> rockchip-linux/master
Fetching projects:  65% (17/26)7), 8.96 MiB | 439.00 KiB/s   
fatal: Couldn't find remote ref refs/heads/rkvideo.00 KiB/s   /s   
error: Cannot fetch gstreamer-rockchip from https://github.com/rockchip-linux/gstreamer-rockchip

    result = cmd.Execute(copts, cargs)
  File "/home/yash/KEPU/repo/subcmds/sync.py", line 855, in Execute
    project.Sync_LocalHalf(syncbuf, force_sync=opt.force_sync)
  File "/home/yash/KEPU/repo/project.py", line 1380, in Sync_LocalHalf
    self._InitWorkTree(force_sync=force_sync, submodules=submodules)
  File "/home/yash/KEPU/repo/project.py", line 2594, in _InitWorkTree
    raise GitError("cannot initialize work tree for " + self.name)
error.GitError: cannot initialize work tree for gstreamer-rockchip

JeffyCN, 2019-09-24:
There's no rkvideo branch here, guess the manifests are out-dated.

Better to use latest release manifests, e.g. rk3399_linux_release.xml/rk3288_linux_release.xml...

closed 2020-08-12 by JeffyCN