rockchip-linux/gstreamer-rockchip #53: How to adjust the mpph264enc encoder bitrate?

santo4ul, 2019-10-20:
Hi, 

`gst-inspect-1.0 mpph264enc` command doesn't show any parameter to adjust the encoder bitrate.

How do I adjust the encoder bitrate?

Thanks,
San

Asazeus, 2019-12-23:
Hi,

I was also faced with this. You need to modify line 97 of gstmpph264enc.c and set value for "rc_cfg.bps_target". This value could be hardcoded or you can add gstreamer parameter and specify it in your pipeline. Now this value calculates from video resolution and framerate.

Sorry about my English.

santo4ul, 2020-02-29:
Hi @Asazeus, thanks for the info. I'll close this ticket.

Thanks,
San

stylesuxx, 2020-03-02:
I really do not think that this issue should be closed. Rather it should be properly implemented. I am a bit baffled that a gstreamer plugin is provided where not even the most basic encoding options can be set.

@hizukiayaka you seem to be the main contributer here - can you tell me if there are any plans on implementing those options?

santo4ul, 2020-03-03:
@stylesuxx I've reopened. I agree that basic parameters should be implemented.

santo4ul, 2020-04-07:
@stylesuxx Thanks for the [commit](https://github.com/stylesuxx/gstreamer-rockchip/commit/195e483a1a8dd6777b232b53bf0adc48baa365f4).

closed 2020-04-07 by santo4ul