rockchip-linux/gstreamer-rockchip #54: mpph264enc0 can't handle caps video/x-raw, format=(string)nv12, width=(int)1920, height=(int)1080, framerate=(fraction)30/1

gounthar, 2019-11-18:
Hi,

I'm trying to combine the output of your ffmpeg fork to gst-launch in order to encode with h.264.

```bash
sudo  ffmpeg  -f v4l2 -pix_fmt nv12 -s 1920x1080 -r 30 -i /dev/video0 -c:v rawvideo -f rawvideo - | gst-launch-1.0 fdsrc ! 'video/x-raw,format=nv12,width=1920,height=1080,framerate=30/1'! mpph264enc ! filesink location=test1.mp4
```

It looks like the rawvideo format is not supported by `mpph264enc`.
What are the supported formats?
Thanks.

JeffyCN, 2019-11-19:
FYI, NV12 not nv12.

gounthar, 2019-11-21:
Thanks. Now, with `NV12`and not `nv12`, I get:

```bash
sudo  ffmpeg  -f v4l2 -pix_fmt nv12 -s 1920x1080 -r 30 -i /dev/video0 -c:v rawvideo -f rawvideo - | gst-launch-1.0 fdsrc ! 'video/x-raw,format=NV12,width=1920,height=1080,framerate=30/1'! mpph264enc ! filesink location=test1.mp4
ffmpeg version 7964b6e2af Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 9.2.0 (GCC)
  configuration: --enable-gpl --enable-version3 --enable-static --disable-shared --enable-small --enable-avisynth --enable-chromaprint --enable-frei0r --enable-gmp --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmodplug --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-librtmp --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libv4l2 --enable-libvo-amrwbenc --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libxml2 --enable-libzmq --enable-libzvbi --enable-lv2 --enable-libmysofa --enable-openal --enable-libdrm --enable-nonfree --enable-libfdk-aac --enable-libbluray --enable-hwaccel='h264_vaapi,hevc_vaapi,vp9_vaapi' --enable-decoder='vp8,vp9' --extra-libs='-lnettle -lhogweed -lgmp' --arch=arm
  libavutil      56. 22.100 / 56. 36.100
  libavcodec     58. 35.100 / 58. 62.100
  libavformat    58. 20.100 / 58. 35.100
  libavdevice    58.  5.100 / 58.  9.101
  libavfilter     7. 40.101 /  7. 66.100
  libswscale      5.  3.100 /  5.  6.100
  libswresample   3.  3.100 /  3.  6.100
  libpostproc    55.  3.100 / 55.  6.100
[video4linux2,v4l2 @ 0xab976e80] The driver does not permit changing the time per frame
[video4linux2,v4l2 @ 0xab976e80] Time per frame unknown
Setting pipeline to PAUSED ...
mpi: mpp version: b212d1b author: Jacob Chen [mpp]: fix mjpeg buffer leak
Pipeline is PREROLLING ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Got EOS from element "pipeline0".
Execution ended after 0:00:00.000671125
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
mpp_thread: MPP_THREAD_STOPPING status set mThread 0xaacd7180
Freeing pipeline ...
```

JeffyCN, 2019-11-29:
Try:
root@jeffy:/# cat /nv12.yuv | gst-launch-1.0 fdsrc ! rawvideoparse format=nv12 width=1920 height=1080 framerate=30/1 ! mpph264enc ! filesink location=test1.mp4

gounthar, 2019-11-29:
Thanks.
It's better, because I don't have the error anymore, but after a few seconds, the board (OrangePi RK3399) hangs.

```
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: rawvideo (NV12 / 0x3231564E), nv12, 1920x1080, 1000k tbr, 1000k tbn, 1000k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> rawvideo (native))
Press [q] to stop, [?] for help
Output #0, rawvideo, to 'pipe:':
  Metadata:
    encoder         : Lavf58.35.100
    Stream #0:0: Video: rawvideo (NV12 / 0x3231564E), nv12, 1920x1080, q=2-31, 746496 kb/s, 30 fps, 30 tbn, 30 tbc
    Metadata:
      encoder         : Lavc58.62.100 rawvideo
h264e_api: h264e_config MPP_ENC_SET_RC_CFG bps 7776000 [7290000 : 8262000]
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
frame=   82 fps=1.9 q=-0.0 size=  249075kB time=00:00:02.73 bitrate=746496.1kbits/s speed=0.065x
```

When the machine is restarted, the result file is empty.

JeffyCN, 2019-12-02:
The hang and reboot should related to board stability, usually about OS code base(related driver version/CPU or DDR's freq & voltage) :(

closed 2019-12-02 by JeffyCN