rockchip-linux/gstreamer-rockchip #59: Error when trying to run Gstreamer pipeline with mppjpegdec

vincent911001, 2020-03-12:
Hi guys,

I have encountered errors when trying to use hardware decoding on RK3399ProD.
The pipeline which produces the issue is as below:
`gst-launch-1.0 v4l2src --gst-debug-level=3 device=/dev/video0 ! image/jpeg,width=640,height=480 ! queue ! jpegparse ! queue ! mppjpegdec ! queue ! videoconvert ! video/x-raw,format=NV12,width=640,height=480 ! queue ! autovideosink`

Besides, software decoding is working which is reproducible with pipeline below:
`gst-launch-1.0 v4l2src ! image/jpeg,width=640,height=480  ! jpegparse ! jpegdec ! videoconvert !  autovideosink`

The error from running the problematic pipeline is as below:
```
Setting pipeline to PAUSED ...
mpi: mpp version: Without VCS info
Pipeline is live and does not need PREROLL ...
0:00:01.069799744 18108   0x55b1e83800 FIXME           videodecoder gstvideodecoder.c:933:gst_video_decoder_drain_out:<mppjpegdec0> Sub-class should implement drain()
Got context from element 'autovideosink0': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";
0:00:01.074859498 18108   0x55b1e838f0 WARN          v4l2bufferpool gstv4l2bufferpool.c:790:gst_v4l2_buffer_pool_start:<v4l2src0:pool:src> Uncertain or not enough buffers, enabling copy threshold
Setting pipeline to PLAYING ...
New clock: GstSystemClock
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
0:00:01.378254538 18108   0x55b1e83800 FIXME           videodecoder gstvideodecoder.c:933:gst_video_decoder_drain_out:<mppjpegdec0> Sub-class should implement drain()

(gst-launch-1.0:18108): GStreamer-CRITICAL **: 21:17:09.065: gst_buffer_peek_memory: assertion 'GST_IS_BUFFER (buffer)' failed

** (gst-launch-1.0:18108): CRITICAL **: 21:17:09.065: gst_is_dmabuf_memory: assertion 'mem != NULL' failed

(gst-launch-1.0:18108): GStreamer-CRITICAL **: 21:17:09.066: gst_buffer_peek_memory: assertion 'GST_IS_BUFFER (buffer)' failed

** (gst-launch-1.0:18108): CRITICAL **: 21:17:09.066: gst_is_dmabuf_memory: assertion 'mem != NULL' failed
0:00:01.488635601 18108   0x55b1e83800 ERROR      mppbarebufferpool gstmppbarebufferpool.c:396:gst_mpp_bare_buffer_pool_fill_frame:<mppjpegdec0:pool:src:allocator> can't fill frame with an invalid buffer
0:00:01.488692475 18108   0x55b1e83800 ERROR             mppjpegdec gstmppjpegdec.c:467:gst_mpp_jpeg_dec_handle_frame:<mppjpegdec0> can't process this frame
0:00:01.553752664 18108   0x55b1e838f0 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<v4l2src0> error: Internal data stream error.
0:00:01.553887704 18108   0x55b1e838f0 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<v4l2src0> error: streaming stopped, reason error (-5)
0:00:01.554116950 18108   0x55b1e838f0 WARN                   queue gstqueue.c:988:gst_queue_handle_sink_event:<queue0> error: Internal data stream error.
0:00:01.554171199 18108   0x55b1e838f0 WARN                   queue gstqueue.c:988:gst_queue_handle_sink_event:<queue0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.475425762
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
mpp_buffer: mpp_group 0x7f4c00ec00 tag unknown caller gst_mpp_jpeg_dec_handle_frame mode internal type ion deinit with 614400 bytes not released
mpp_buffer: 
dumping buffer group 0x7f4c00ec00 id 3 from put_group
mpp_buffer: mode internal
mpp_buffer: type ion
mpp_buffer: limit size 0 count 0
mpp_buffer: used buffer count 1
mpp_buffer: buffer 0x7f4c00eee0 fd   34 size     614400 ref_count   1 discard 0 caller gst_mpp_jpeg_dec_handle_frame
mpp_buffer: unused buffer count 0
Setting pipeline to NULL ...
Freeing pipeline ...
mpp_buffer: ~MppBufferService cleaning leaked group
mpp_buffer: ~MppBufferService cleaning leaked buffer
free(): invalid pointer
Aborted (core dumped)
```

Thanks.

dmacattack, 2020-03-13:
I have had problems with this plugin also. I did not have the same issues as you, mine didnt crash, but it was very slow. I have found the libmpp library is very specific to the jpeg frames. For my purposes I ended up patching libmpp to suit the frames output by my specific camera. Theres a loop in jpegd_find_marker() that loops looking for restart markers, if it doesnt find them it searches the entire frame.
``` patch
-        if ((v == 0xff) && (v2 >= 0xc0) && (v2 <= 0xfe) && buf_ptr < buf_end) {
-            val = *buf_ptr++;
+
+        if ((v == 0xff) && (v2 >= 0xc0) && (v2 <= 0xfe) && buf_ptr < buf_end) 
+        {
+
+            if ( (v2 >= 0xd0) && (v2 <= 0xd7) )
+            {
+                // EDIT: skip restart markers, since theyre not implemented
+                continue;
+            }
+
+            val = *buf_ptr++; // INSP: val is the marker (0xd8 (SOI), 0xe0(APP0), ... 0xd9 (EOI) )
             goto found;
-        } else if ((v == 0x89) && (v2 == 0x50)) {
-            mpp_log("input img maybe png format,check it\n");
-        }
-        skipped++;
+        } 
```

good luck. I'd suggest sprinkling this library with mpp_logs and examining whats happening before your crash.

JeffyCN, 2020-08-11:
could be fixed by:
commit 0595286b4589bd763c67f5cae4d9ee686f20999b
Author: Jeffy Chen <jeffy.chen@rock-chips.com>
Date:   Tue Nov 12 16:49:25 2019 +0800

    rockchipmpp: mppjpegdec: Correct input packet's size and length

    Change-Id: Ibee81a7b7de4afd986fc15bf6765bc2648438db3
    Signed-off-by: Jeffy Chen <jeffy.chen@rock-chips.com>

closed 2020-08-11 by JeffyCN