rockchip-linux/gstreamer-rockchip #60: Error when trying to use hardware decoding on camera stream with mppvideodec

vincent911001, 2020-03-12:
Hi guys,

While mppvideodec is working when it is used in video playback using `filesrc`, it is not working when `filesrc` is substituted with `v4l2src`.

Working pipeline:
`gst-launch-1.0 filesrc location=product.mp4 ! qtdemux ! h264parse ! mppvideodec ! kmssink`

Problematic pipeline: (have verified using v4l2-ctl that this camera is able to produce 1080p video stream in h264)
```
gst-launch-1.0 v4l2src ! video/x-h264,width=1920,height=1080 ! h264parse ! mppvideodec ! kmssink
Setting pipeline to PAUSED ...
mpi: mpp version: Without VCS info
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpp: deprecated block control, use timeout control instead

** (gst-launch-1.0:18565): CRITICAL **: 21:50:02.691: gst_mpp_allocator_start: assertion 'size != 0' failed
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
Execution ended after 0:00:00.366435265
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
Any ideas?

vincent911001, 2020-03-12:
Additional log by enabling gst-debug-level=1

```
Setting pipeline to PAUSED ...
mpi: mpp version: Without VCS info
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

** (gst-launch-1.0:18897): CRITICAL **: 22:16:01.936: gst_mpp_allocator_start: assertion 'size != 0' failed
0:00:01.877005082 [334m18897[00m   0x55a12825e0 [31;01mERROR  [00m [00m    mppdecbufferpool gstmppdecbufferpool.c:112:gst_mpp_dec_buffer_pool_start:<mppvideodec0:pool:src>[00m we received 0 buffer, we want at least 22
0:00:01.877151496 [334m18897[00m   0x55a12825e0 [31;01mERROR  [00m [00m          bufferpool gstbufferpool.c:564:gst_buffer_pool_set_active:<mppvideodec0:pool:src>[00m start failed
0:00:01.877204579 [334m18897[00m   0x55a12825e0 [31;01mERROR  [00m [00m         mppvideodec gstmppvideodec.c:663:gst_mpp_video_dec_handle_frame:<mppvideodec0>[00m Unable to activate the pool
0:00:01.877241037 [334m18897[00m   0x55a12825e0 [31;01mERROR  [00m [00m         mppvideodec gstmppvideodec.c:693:gst_mpp_video_dec_handle_frame:<mppvideodec0>[00m can't process this frame
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason error (-5)
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
mpp: deprecated block control, use timeout control instead
Execution ended after 0:00:00.456789851
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

JeffyCN, 2020-08-12:
please try newest version, if still repro, you can try to add logs in:
gstmppvideodec.c:gst_mpp_video_frame_to_info

the buffer size is calculated here, and should not be zero.

JeffyCN, 2020-08-25:
please re-open if failed in the newest version.

closed 2020-08-25 by JeffyCN