rockchip-linux/gstreamer-rockchip #62: mppjpegdec only 11fps headless

coudy, 2020-04-14:
JPEG decoding using mppjpegdec is very slow (<12fps), much slower than jpegdec when followed by other non-RK elements, see commands below. 

1. I suspect the bottleneck is copying buffers from the VPU memory but don't understand the internals well enough. Can someone please verify that this is indeed the bottleneck?
2. If the above is correct, can I do more operations directly in DMA, in particular video conversion, transcoding, RTP streaming or saving to a file?
3. In general, is it possible to get 1080p @ 30fps decoding in a headless application?

More info:
- Board: Khadas Edge-V (RK3399)
- gstreamer: 1.14.5
- gstreamer-rockchip: c86aced
- mpp: https://github.com/rockchip-linux/mpp/commit/e26a59a676da21af540e499d071690bc7047e192

```
khadas@Khadas:~/ng$ gst-launch-1.0 v4l2src io-mode=2 num-buffers=30 device=/dev/video0 ! 'image/jpeg,width=1920,height=1080,framerate=30/1' ! jpegparse ! mppjpegdec ! videoconvert ! avimux ! fpsdisplaysink video-sink="filesink location=1080p_raw_mppjpegdec2.avi" sync=false text-overlay=false -ve
Setting pipeline to PAUSED ...
mpp[2827]: mpp_info: mpp version: e26a59a6 author: sayon.chen    2020-04-01 [mpp_enc_v2]: Fix reenc path task_length asssert
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFileSink:filesink0: sync = false
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstJpegParse:jpegparse0.GstPad:sink: caps = image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstJpegParse:jpegparse0.GstPad:src: caps = image/jpeg, parsed=(boolean)true, format=(string)UYVY, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
mpp[2827]: mpp_rt: NOT found ion allocator
mpp[2827]: mpp_rt: found drm allocator
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:sink: caps = image/jpeg, parsed=(boolean)true, format=(string)UYVY, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstMppJpegDec:mppjpegdec0.GstPad:src: caps = video/x-raw, format=(string)NV16, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)bt709, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, format=(string)YUY2
/GstPipeline:pipeline0/GstAviMux:avimux0.GstPad:video_0: caps = video/x-raw, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, format=(string)YUY2
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, format=(string)NV16, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)bt709, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstAviMux:avimux0.GstPad:src: caps = video/x-msvideo
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-msvideo
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFileSink:filesink0.GstPad:sink: caps = video/x-msvideo
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-msvideo
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFileSink:filesink0: sync = false
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 6, dropped: 0, current: 11.85, average: 11.85
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 12, dropped: 0, current: 11.14, average: 11.48
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 18, dropped: 0, current: 11.40, average: 11.46
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 24, dropped: 0, current: 11.47, average: 11.46
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 30, dropped: 0, current: 11.46, average: 11.46
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 36, dropped: 0, current: 11.48, average: 11.46
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 42, dropped: 0, current: 11.46, average: 11.46
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 48, dropped: 0, current: 11.47, average: 11.46
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 54, dropped: 0, current: 11.49, average: 11.47
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 60, dropped: 0, current: 11.46, average: 11.47
Got EOS from element "pipeline0".
Execution ended after 0:00:05.873341010
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

```
khadas@Khadas:~/ng$ gst-launch-1.0 v4l2src io-mode=2 num-buffers=30 device=/dev/video0 ! 'image/jpeg,width=1920,height=1080,framerate=30/1' ! jpegparse ! jpegdec ! videoconvert ! avimux ! fpsdisplaysink video-sink="filesink location=1080p_raw_jpegdec2.avi" sync=false text-overlay=false -ve
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFileSink:filesink0: sync = false
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstJpegParse:jpegparse0.GstPad:sink: caps = image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = image/jpeg, width=(int)1920, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:7:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstJpegParse:jpegparse0.GstPad:src: caps = image/jpeg, parsed=(boolean)true, format=(string)UYVY, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstJpegDec:jpegdec0.GstPad:sink: caps = image/jpeg, parsed=(boolean)true, format=(string)UYVY, width=(int)1920, height=(int)1080, framerate=(fraction)30/1
[... cut ...]
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstFileSink:filesink0: sync = false
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 22, dropped: 0, current: 41.64, average: 41.64
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0: last-message = rendered: 54, dropped: 0, current: 63.95, average: 52.49
Got EOS from element "pipeline0".
Execution ended after 0:00:01.527107054
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

JeffyCN, 2020-07-15:
There used to be 3 patches for this issue, but they are out-dated @HermanChen:
https://github.com/rockchip-linux/rk-rootfs-build/blob/master/packages-patches/mpp/release_20171218/0001-WIP-allow-decode-simple-for-jpeg.patch
https://github.com/rockchip-linux/rk-rootfs-build/blob/master/packages-patches/gstreamer-rockchip/release_20171218/0002-DEBIAN-ONLY-rockchipmpp-allow-jpeg.patch
https://github.com/rockchip-linux/rk-rootfs-build/blob/master/packages-patches/gstreamer-rockchip/release_20171218/0004-DEBIAN-ONLY-rockchipmpp-always-use-simple-io-mode.patch

or try this patch:
https://filebin.net/2hbyw7x45m747tzu/0001-rockchipmpp-mppjpegdec-Convert-to-async-decoding.patch?t=4cyby1f8

closed 2020-08-11 by JeffyCN

kadirbeytorun, 2021-02-27:
@JeffyCN The links you provided are broken, could you look into this again?
I am having the same issue on rk1808

JeffyCN, 2021-02-27:
already merged, try newest version