rockchip-linux/gstreamer-rockchip #63: mpph264enc produces no output after lots of failed assertions

coudy, 2020-04-14:
Having built the latest gstreamer-rockchip against the latest mpp, I'm getting an error in a simple encoding pipeline on RK3399. I've tried a raw I420 file input as well as a YUY2 webcam input:

```
$ gst-launch-1.0  v4l2src num-buffers=5 ! 'video/x-raw,framerate=30/1,width=640,height=480' ! mpph264enc ! h264parse ! fakesink
Setting pipeline to PAUSED ...
mpp[17309]: mpp_info: mpp version: 50a96555 author: Herman Chen   2020-04-10 [test]: update sample for fbc support
mpp[17309]: mpp_rt: NOT found ion allocator
mpp[17309]: mpp_rt: found drm allocator
mpp[17309]: rc: rc_init using rc impl default
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
mpp[17309]: h264e_api_v2: h264e_proc_cfg MPP_ENC_SET_RC_CFG bps 1152000 [1080000 : 1224000] fps [30:30] gop 30
mpp[17309]: h264e_api_v2: h264e_proc_cfg MPP_ENC_SET_PREP_CFG w:h [640:480] stride [1280:480]
mpp[17309]: h264e_sps: set level to 2.2
mpp[17309]: mpp_enc_v2: Please use MPP_ENC_GET_HDR_SYNC instead of unsafe MPP_ENC_GET_EXTRA_INFO
mpp[17309]: mpp_enc_v2: NOTE: MPP_ENC_GET_HDR_SYNC needs MppPacket input
mpp[17309]: mpp_enc_v2: set_rc_cfg mode: cbr
mpp[17309]: mpp_enc_v2: set_rc_cfg bps: [1080000 : 1152000 : 1224000]
mpp[17309]: mpp_enc_v2: set_rc_cfg fps: fix [30/1] -> fix [30/1]
mpp[17309]: mpp_enc_v2: set_rc_cfg gop: i [30] v [0]
mpp[17309]: mpp_enc_v2: Assertion hal_task->length == mpp_packet_get_length(packet) failed at mpp_enc_thread:686
mpp[17309]: hal_h264e_vepu_v2: warnning: input buffer size 0x96000 is smaller then required size 0x12c000
mpp[17309]: mpp_enc_v2: Assertion hal_task->length == mpp_packet_get_length(packet) failed at mpp_enc_thread:732
mpp[17309]: mpp_enc_v2: Assertion hal_task->length == mpp_packet_get_length(packet) failed at mpp_enc_thread:738
mpp[17309]: hal_h264e_vepu_v2: warnning: input buffer size 0x96000 is smaller then required size 0x12c000
mpp[17309]: mpp_enc_v2: Assertion hal_task->length == mpp_packet_get_length(packet) failed at mpp_enc_thread:686
mpp[17309]: hal_h264e_vepu_v2: warnning: input buffer size 0x96000 is smaller then required size 0x12c000
mpp[17309]: mpp_enc_v2: Assertion hal_task->length == mpp_packet_get_length(packet) failed at mpp_enc_thread:686
mpp[17309]: hal_h264e_vepu_v2: warnning: input buffer size 0x96000 is smaller then required size 0x12c000
mpp[17309]: mpp_enc_v2: Assertion hal_task->length == mpp_packet_get_length(packet) failed at mpp_enc_thread:686
mpp[17309]: hal_h264e_vepu_v2: warnning: input buffer size 0x96000 is smaller then required size 0x12c000
mpp[17309]: mpp_enc_v2: Assertion hal_task->length == mpp_packet_get_length(packet) failed at mpp_enc_thread:686
mpp[17309]: hal_h264e_vepu_v2: warnning: input buffer size 0x96000 is smaller then required size 0x12c000
mpp[17309]: mpp_enc_v2: Assertion hal_task->length == mpp_packet_get_length(packet) failed at mpp_enc_thread:686
mpp[17309]: hal_h264e_vepu_v2: warnning: input buffer size 0x96000 is smaller then required size 0x12c000
ERROR: from element /GstPipeline:pipeline0/GstH264Parse:h264parse0: No valid frames found before end of stream
Additional debug info:
gstbaseparse.c(1329): gst_base_parse_sink_event_default (): /GstPipeline:pipeline0/GstH264Parse:h264parse0
Execution ended after 0:00:00.476003530
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

More info:
- Board: Khadas Edge-V (RK3399)
- gstreamer: 1.14.5
- gstreamer-rockchip: c86aced
- mpp: rockchip-linux/mpp@50a9655531b65d54c070a4d721b9a3ca14537e43

DOT output from GStreamer:
[image]

JeffyCN, 2020-07-15:
maybe try this:
https://github.com/rockchip-linux/gstreamer-rockchip/commit/c9a9550abac2b5ffce11b5d0ec68ac4b47e7909f

JeffyCN, 2020-07-17:
fixed

closed 2020-07-17 by JeffyCN