rockchip-linux/gstreamer-rockchip #66: mppvideodec: When changing pipeline state from READY to PLAYING, there is ~ 0.2 seconds delay before playback starts

whoman321, 2020-07-29:
mppvideodec: When changing pipeline state from READY to PLAYING, there is approximately 0.2 seconds delay before playback begins

This delay causes 2 problems:
1. when attempting to change source, the start of playback of second source is delayed.
2. during this delay, the last frame from previous source is frozen on the sink / screen.

Attached is the python example.
[example3.zip](https://github.com/rockchip-linux/gstreamer-rockchip/files/4991843/example3.zip)

GST_DEBUG=3 output....

This is where the delay occurs:

0:00:01.761520179  1872   0x5593664ed0 FIXME           videodecoder gstvideodecoder.c:945:gst_video_decoder_drain_out:<decoder> Sub-class should implement drain()
mpp[1872]: mpp_rt: NOT found ion allocator
mpp[1872]: mpp_rt: found drm allocator
0:00:01.773728557  1872   0x5593664ed0 FIXME           videodecoder gstvideodecoder.c:945:gst_video_decoder_drain_out:<decoder> Sub-class should implement drain()
mpp[1872]: mpp: deprecated block control, use timeout control instead
mpp[1872]: mpp: deprecated block control, use timeout control instead
0:00:01.947462786  1872   0x5593665d40 WARN            kmsallocator gstkmsallocator.c:556:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22

**As you can see by the timestamps, the freeze happens between the FIXME and the WARN debug lines**

This problem / delay does not occur on intel with VAAPI hw decoder.

For our product, we need playback to begin without any delay/latency.

Here is the entire DEBUG output (below is example python):
$ GST_DEBUG=3 python example3.py 

setting STATE to READY
mpp[1872]: mpp_info: mpp version: df0a7f02 author: Herman Chen   2020-07-28 [h264e]: Add poc type / poc_lsb / frm_num setup

changing filesrc location
filesrc location has been changed.  Setting STATE back to PLAYING
0:00:01.750940187  1872   0x5593875600 WARN                 basesrc gstbasesrc.c:3491:gst_base_src_start_complete:<source> pad not activated yet

0:00:01.753216953  1872   0x5593664ed0 WARN                 qtdemux qtdemux_types.c:223:qtdemux_type_get: unknown QuickTime node type iods
0:00:01.753662915  1872   0x5593664ed0 WARN                 qtdemux qtdemux_types.c:223:qtdemux_type_get: unknown QuickTime node type gsst
0:00:01.753813999  1872   0x5593664ed0 WARN                 qtdemux qtdemux_types.c:223:qtdemux_type_get: unknown QuickTime node type gstd
0:00:01.754073584  1872   0x5593664ed0 WARN                 qtdemux qtdemux.c:2899:qtdemux_parse_trex:<demux> failed to find fragment defaults for stream 1
0:00:01.754635047  1872   0x5593664ed0 WARN                 qtdemux qtdemux.c:2899:qtdemux_parse_trex:<demux> failed to find fragment defaults for stream 2
0:00:01.755648304  1872   0x5593664ed0 WARN                 basesrc gstbasesrc.c:2400:gst_base_src_update_length:<source> processing at or past EOS
Received new pad 'video_0' from 'demux'
video/x-h264
Link succeeded (type 'video/x-h264')
0:00:01.761520179  1872   0x5593664ed0 FIXME           videodecoder gstvideodecoder.c:945:gst_video_decoder_drain_out:<decoder> Sub-class should implement drain()
mpp[1872]: mpp_rt: NOT found ion allocator
mpp[1872]: mpp_rt: found drm allocator
0:00:01.773728557  1872   0x5593664ed0 FIXME           videodecoder gstvideodecoder.c:945:gst_video_decoder_drain_out:<decoder> Sub-class should implement drain()
mpp[1872]: mpp: deprecated block control, use timeout control instead
mpp[1872]: mpp: deprecated block control, use timeout control instead
0:00:01.947462786  1872   0x5593665d40 WARN            kmsallocator gstkmsallocator.c:556:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22
0:00:02.030347830  1872   0x5593665d40 WARN            kmsallocator gstkmsallocator.c:556:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22
0:00:02.072035167  1872   0x5593665d40 WARN            kmsallocator gstkmsallocator.c:556:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22

JeffyCN, 2020-08-05:
base on the logs, it should be blocked at the buffer allocation in gst_mpp_allocator_drm_buf(), you can try to add logs there to confirm.

and for "the last frame frozen", it should be a sink side issue(clear screen when sink stopped).

whoman321, 2020-08-06:
Thank you for the information.
 
I placed debug timestamps throughout **gst_mpp_allocator_drm_buf**, and I have determined where the delay is caused.

this **for** loop within **gst_mpp_allocator_drm_buf**:
```
  for (gint i = 0; i < count; i++) {
    /*
     * Create MppBuffer from Rockchip Mpp
     * included mvc data
     */
    if (mpp_buffer_get (group, &temp_buf[i], size)) {
      GST_ERROR_OBJECT (allocator, "allocate internal buffer %d failed", i);
      goto error;
    }
  }
```
Specifically, it's the **if** statement:
```
   if (mpp_buffer_get (group, &temp_buf[i], size)) {
```
The **for** loop causes this **if** statement to execute 22 times,
and each time this if statement is called, there is approximately 0.01 seconds delay
Because this line is executed 22 times within the for loop, it adds up to approximately 0.22 seconds delay total.

I'm not sure where to go from here.  How can we fix this?
I see this comment within the function:
```
 /* FIXME the rockchip mpp should support DRM type properly */
```
Could this be causing the problem / delay?

Thanks very much for your help

JeffyCN, 2020-08-06:
hmm, that is not a bug actually, the decoder needs 22 output frames:
https://github.com/rockchip-linux/gstreamer-rockchip/blob/master/gst/rockchipmpp/gstmppvideodec.c#L38

not sure if we can reduce the buffers @HermanChen

and mpp is allocating buffers from drm now, which do need time.

whoman321, 2020-08-06:
Thanks for your reply.

I understand the 22 frame minimum for rockchip MPP (for h264).

The problem is not the number of frames...  the problem is how much time it takes for each frame.

Each frame allocation is taking a very long time (about 0.01 seconds) compared to many other platforms (including other ARM).

Why does MPP (specifically **mpp_buffer_get**) take so much time (0.01 seconds) for each frame to be buffered?

It should be closer to 0.001 - 0.003 per frame.

Perhaps there is a problem with **mpp_buffer_get**?

I very much appreciate your time.

JeffyCN, 2020-08-06:
like i said "mpp is allocating buffers from drm now, which do need time", and the time is related to the buffer size.

whoman321, 2020-08-06:
Do you believe this 0.20 second delay when buffering h264 on rockchip is working correctly (no delay)?

Thanks for the kernel drm info... I will take a look at  drm_gem_get_pages, however since this problem does not affect other platforms, I don't know how kernel drm would be the blocked.

Am I not understanding correctly?

I will check  drm_gem_get_pages.  Thank you for your advice and help.

JeffyCN, 2020-08-06:
believe or not, the drm does that.

i'm not sure about other platforms, but if they are using the same way (drm_gem_get_pages), it should be the same.

anyway, there are multiple ways to allocate buffers, for example kernel's CMA or ION. i'm tried CMA, it's quick, but breaks the display. and i believe there're some documents under docs/ about how to enable ION.

whoman321, 2020-08-06:
Okay sounds good... I'll try these things:
1.  I will investigate if other platforms without delay use drm_gem_get_pages
2.  I will research how to enable/use ION
3.  I will test using ramdisk to ensure this is not a physical EMMC read delay

one question... for ION, I see this:
```
Currently it is designed for ion buffer on Android and drm buffer on Linux
```
http://opensource.rock-chips.com/wiki_Mpp

I assumed this means that ION will not work on Linux.  Can ION work on Linux?

JeffyCN, 2020-08-06:
i think it should work, that is just a driver for managing dma buffers.

FYI i tried a simple loop to allocate buffers, and perf tool got:
               |--99.81%-- gst_mpp_allocator_drm_buf
               |          |
               |          |--77.35%-- mpp_buffer_get_with_tag
          |          |          |                     |--73.92%-- rockchip_gem_create_object
          |          |          |                     |          |--66.07%-- rockchip_gem_get_pages
               |          |          |          |          |          |                     |          |          |                     |          |          |     
               |          |          |          |          |          |                     |          |          |                     |          |          |--47.63%-- drm_gem_get_pages
     |--47.59%-- shmem_read_mapping_page_gfp
               |          |          |          |          |          |                     |          |          |                     |          |          |          |          |
               |          |          |          |          |          |                     |          |          |                     |          |          |          |          |--19.90%-- shmem_getpage_gfp
               |          |          |          |          |          |                     |          |          |                     |          |          |          |          |          |

the gst_mpp_allocator_drm_buf's 99.81%-77.35% is freeing buffers, which is unrelated.

the rockchip_gem_get_pages's 66.07%-47.63% should due to:
commit e81c6035cd92549c354d5daa52bd31e9394c9cd8
Author: Simon <xxm@rock-chips.com>
Date:   Tue Jan 17 09:23:19 2017 +0800

    drm/rockchip: gem: reorder pages if page chunk less than 8

    Change-Id: I03a91d2f9c017086b3cb35edeaf6b7913b147b9b
    Signed-off-by: Simon <xxm@rock-chips.com>

whoman321, 2020-08-06:
Wow thank you! I really appreciate your efforts!

Could you provide a link to this commit?
(I'm having problems finding it with Change-Id)

HermanChen, 2020-08-06:
About the buffer count the max reference frame in H.264 is 16. Plus current decoding frame. And 2 deinterlace frame queue and 3 buffer for display queue. Then the max buffer number can reach 22.

JeffyCN, 2020-08-06:
to use ion:
1/ enable kernel configs: CONFIG_ION and CONFIG_ION_ROCKCHIP

2/ enable kernel dts:
+++ b/arch/arm64/boot/dts/rockchip/rk3399-linux.dtsi
@@ -55,6 +55,14 @@
                #size-cells = <2>;
                ranges;

+               /* global autoconfigured region for contiguous allocations */
+               linux,cma {
+                       compatible = "shared-dma-pool";
+                       reusable;
+                       size = <0x0 0x8000000>;
+                       linux,cma-default;
+               };
+
@@ -151,6 +159,19 @@
                dma-coherent;
                status = "okay";
        };
+
+       ion {
+               compatible = "rockchip,ion";
+               #address-cells = <1>;
+               #size-cells = <0>;
+
+               cma-heap {
+                       reg = <0x00000000 0x02000000>;
+               };
+
+               system-heap {
+               };
+       };
 };

3/ disable mpp check:
diff --git a/osal/mpp_runtime.cpp b/osal/mpp_runtime.cpp
index 05e6db91..94244cb7 100644
--- a/osal/mpp_runtime.cpp
+++ b/osal/mpp_runtime.cpp
@@ -105,7 +105,7 @@ MppRuntimeService::MppRuntimeService()

     // If both ion and drm is enabled detect allocator in dts to choose one
     // TODO: When unify dma fd kernel is completed this part will be removed.
-    if (allocator_valid[MPP_BUFFER_TYPE_ION] &&
+    if (0 && allocator_valid[MPP_BUFFER_TYPE_ION] &&

0:00:00.132185211  2672   0x55b3432770 FIXME           videodecoder gstvideodecode
r.c:945:gst_video_decoder_drain_out:<decoder> Sub-class should implement drain()
mpp[2672]: mpp_rt: found ion allocator        <--- found ion
...
mpp[2672]: mpp_ion: using ion heap ION_HEAP_TYPE_SYSTEM       <-------- use ion
...
0:00:00.143759969  2672   0x55b3432770 FIXME           videodecoder gstvideodecode
r.c:945:gst_video_decoder_drain_out:<decoder> Sub-class should implement drain()
mpp[2672]: mpp: deprecated block control, use timeout control instead
mpp[2672]: mpp: deprecated block control, use timeout control instead
0:00:00.187407537  2672   0x55b3433320 WARN            kmsallocator gstkmsallocato <----- 50ms

whoman321, 2020-08-07:
Wow!  50ms would be great! 
Thank you very very much for taking the time to make the ION patches
I will test out the patches on Monday, and will reply with my results.

whoman321, 2020-08-12:
I've tried this patch but mpp is still choosing DRM.

I'm using RK3328, so I did patch my dtsi by hand, based on your RK3399 patch
I also added the 2 lines in my defconfig to enable CONFIG_ION and CONFIG_ION_ROCKCHIP

enable 'ion' is successful:

```
dmesg:

[    0.258699] rockchip ion: success to create - cma-heap
[    0.259212] rockchip ion: success to create - system-heap
[    1.381576] rga2 ff390000.rga: rga ion client create success!
```
```
ls -l /dev/ion:

crw------- 1 root root 10, 63 Aug 12 09:46 /dev/ion
```
```
ls -l /proc/device-tree/ion/:

total 0
-r--r--r-- 1 root root  4 Aug 12 19:03 #address-cells
-r--r--r-- 1 root root  4 Aug 12 19:03 #size-cells
drwxr-xr-x 2 root root  0 Aug 12 19:03 cma-heap
-r--r--r-- 1 root root 13 Aug 12 19:03 compatible
-r--r--r-- 1 root root  4 Aug 12 19:03 name
drwxr-xr-x 2 root root  0 Aug 12 19:03 system-heap
```

for mpp, I have applied the patch:
```
cat mpp_runtime.cpp |grep '0 &&':

if (0 && allocator_valid[MPP_BUFFER_TYPE_ION] &&
```
however, gstreamer does not choose ion:

```
0:00:00.899501073  1068   0x557f22bb70 FIXME           videodecoder gstvideodecoder.c:945:gst_video_decoder_drain_out:<mppvideodec0> Sub-class should implement drain()
mpp[1068]: mpp_rt: NOT found ion allocator
mpp[1068]: mpp_rt: found drm allocator
0:00:00.920647666  1068   0x557f22bb70 FIXME           videodecoder gstvideodecoder.c:945:gst_video_decoder_drain_out:<mppvideodec0> Sub-class should implement drain()
mpp[1068]: mpp: deprecated block control, use timeout control instead
mpp[1068]: mpp: deprecated block control, use timeout control instead
0:00:01.109727698  1068   0x7f70015de0 WARN            kmsallocator gstkmsallocator.c:556:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22
```
is there some extra step to enable ion within mpp?

HermanChen, 2020-08-13:
emm, the ion heap detection may fail.

HermanChen, 2020-08-13:
The /dev/ion can only be accessed by root?
Can you chmod?

whoman321, 2020-08-13:
Yes that solved the problem.
Now Ion is working and only 50ms buffer delay.  Very fast.

Thank you so much @JeffyCN 
Thank you so much @HermanChen 

Very much appreciate your help

closed 2020-08-13 by whoman321