rockchip-linux/gstreamer-rockchip #69: mppvideodec: dma_duf: very big memory leak

whoman321, 2020-08-17:
There is a very big memory leak when using mppvideodec.
This memory leak is happening with both buffer types (DRM and ION)

With Gstreamer, the correct way to change the source of a running pipeline is:

1.  Change pipeline state to READY
2.  Change source (filesrc location paramter)
3.  Change pipeline state to PLAYING

The problem is that dma_buf does not release the memory when the source is changed.
This causes 1 big problem:

96MB more RAM is allocated each time the source is changed.  It's never released.

This means that if I change the source 10 times, I lose ~1GB of RAM.

How to reproduce:
Note:  open 2 terminal sessions:
Terminal 1: Gstreamer session
Terminal 2: for memory debug/monitoring

1.  Terminal 1: Launch python example
2.  Terminal 2: Check dma_buf: **sudo cat /sys/kernel/debug/dma_buf/bufinfo**
```Total 29 objects, 44916736 bytes```
3.  Terminal 1: Press '1' or '2' to change the source
4.  Terminal 2: Check dma_buf: **sudo cat /sys/kernel/debug/dma_buf/bufinfo**

You will notice that buffer/cache usage increases ~100mb
You will notice that dma_buf objects increase from 29 to 58

5.  Terminal 1: Press '1' or '2' to change the source
6.  Terminal 2: Check dma_buf: **sudo cat /sys/kernel/debug/dma_buf/bufinfo**
```Total 58 objects, 141197312 bytes```

You will notice that buffer/cache usage increases ~96mb (none released)
You will notice that dma_buf objects increase from 58 to 87 (none released)

The result:  After changing the source 15 times, all RAM is unavailable, and the program crashes.

Note:
1.  This problem does not happen with sw decoder (avdec_h264)
2.  This problem does not happen with intel hw decoder (vaapidec)

We need to be able to change the source unlimited times for our product.

How can we fix this leak?

python example:
```
import tty
import sys
import termios
import gi
import subprocess
import os, signal
import time
from time import sleep
gi.require_version('Gst', '1.0')
gi.require_version('GLib', '2.0')
gi.require_version('GObject', '2.0')
from gi.repository import Gst, GObject, GLib
GObject.threads_init()
Gst.init(sys.argv)

#GST PIPELINE
#GST - create the elements
source = Gst.ElementFactory.make("filesrc", "source")
demux = Gst.ElementFactory.make("qtdemux", "demux")
parse = Gst.ElementFactory.make("h264parse", "parse")
decoder = Gst.ElementFactory.make("mppvideodec", "decoder")
video_convert = Gst.ElementFactory.make("videoconvert", "videoconvert")
sink = Gst.ElementFactory.make("kmssink", "sink")

#GST - create the empty pipeline
p1 = Gst.Pipeline.new("test-pipeline")

if not p1 or not source or not sink:
	print("ERROR: Not all elements could be created")
	sys.exit(1)

#GST - add and link elements in pipeline
p1.add(source)
p1.add(demux)
p1.add(parse)
p1.add(decoder)
p1.add(video_convert)
p1.add(sink)

if not source.link(demux):
	print("ERROR: Could not link source to demux")
	sys.exit(1)
#if not demux.link(parse):
#	print("ERROR: Could not link demux to parse")
#		sys.exit(1)
if not parse.link(decoder):
	print("ERROR: Could not link parse to decoder")
	sys.exit(1)
if not decoder.link(video_convert):
	print("ERROR: Could not link decoder to video_convert")
#		sys.exit(1)
if not video_convert.link(sink):
	print("ERROR: Could not link scaler to sink")
	sys.exit(1)

# set the source element's location property
source.set_property("location", "/video/g62.mp4")

ret = p1.get_state(1)
print ret.state

#GST - handler for the pad-added signal
def demux_on_pad_added(src, new_pad):
	if new_pad.get_name() == "video_0":
		print(
			"Received new pad '{0:s}' from '{1:s}'".format(
				new_pad.get_name(),
				src.get_name()))

		# attempt the link
		sink_pad = parse.get_static_pad("sink")
		new_pad_caps = new_pad.get_current_caps()
		new_pad_struct = new_pad_caps.get_structure(0)
		new_pad_type = new_pad_struct.get_name()
		print(new_pad_type)
		ret = new_pad.link(sink_pad)
		if not ret == Gst.PadLinkReturn.OK:
			print("Type is '{0:s}' but link failed".format(new_pad_type))
		else:
			print("Link succeeded (type '{0:s}')".format(new_pad_type))
		return
	return

# connect to the pad-added signal
demux.connect("pad-added", demux_on_pad_added)

def dispose_src_cb(src):
    src.set_state(Gst.State.NULL)

p1.set_state(Gst.State.PLAYING)

#GST - Puase playback, set "location", and begin playback
def PlayVideo1():
#	GST stop pipeline
        p1.set_state(Gst.State.READY)
#       GLib.idle_add(dispose_src_cb, decoder)

#	GST set filesrc 'location'
        source.set_property('location', '/video/test1.mp4')
	
#	GST start pipeline
#	decoder.sync_state_with_parent()
        p1.set_state(Gst.State.PLAYING)

#	GST set filesrc 'location'
        print "changing filesrc location"
        source.set_property('location', '/video/test1.mp4')
	print "filesrc location has been changed.  Setting STATE back to PLAYING"

#	GST start pipeline
#        decoder.sync_state_with_parent()
        p1.set_state(Gst.State.PLAYING)
        ret = p1.get_state(1)
        print "current state:"
        print ret.state

	
def PlayVideo2():
#	GST stop pipeline
        p1.set_state(Gst.State.READY)
#       GLib.idle_add(dispose_src_cb, decoder)

#	GST set filesrc 'location'
        source.set_property('location', '/video/test2.mp4')
	
#	GST start pipeline
#	decoder.sync_state_with_parent()
        p1.set_state(Gst.State.PLAYING)

#KEYBOARD - detect key press, and launch GST PlayVideo functions
orig_settings = termios.tcgetattr(sys.stdin)

tty.setcbreak(sys.stdin)
x = 0
while x != chr(27): # ESC
    x=sys.stdin.read(1)[0]
    print("You pressed", x)
    if x == "1":
        PlayVideo1()
    if x == "2":
        PlayVideo2()
    if x == "x":
        exit()
		
termios.tcsetattr(sys.stdin, termios.TCSADRAIN, orig_settings) 
```

whoman321, 2020-08-17:
The decoder (mppvideodec) should release the dma_buf references when state is changed to READY or NULL

Question:
How can I patch mppvideodec to release the allocated dma_buf when state is changed to NULL or READY? 

Any help would be very much appreciated.

JeffyCN, 2020-08-21:
Hi,

This is interesting, there's a simple way to repro it, just attach gdb and add breakpoint at mpp_device_deinit, then check /sys/kernel/debug/dri/0/mm_dump when quiting gst-play-1.0, the drm dma buf would be free-ed as soon as we close the vpu device.

It looks like when stop decoder and resetting mpp ctx, the userspace indeed free-ed all the buffers, but there still are some references in the vcodec driver. adding logs in kernel/drivers/video/rockchip/vcodec/vcodec_iommu_drm.c's dma_buf_get/put confirmed that.

@HermanChen please help to check, we should be able to release all buffers when resetting mpp ctx, or we would have to destroy and recreate mpp stuffs when stop/start decoding

HermanChen, 2020-08-21:
Does the changing source mean change the resolution of input video?
Can you try use video with same size as the source and change them?
Mpp support the change on source with different resolution. But the buffer group of decoder should be reset on each time the source change.

whoman321, 2020-08-21:
even with the same exact video file, the problem occurs.  

no source change necessary to cause this memory leak.

the memory leak occurs just by setting pipeline from PLAYING to READY to PLAYING

when pipeline is set to READY after PLAYING, it should release

JeffyCN, 2020-08-21:
Hi @whoman321 ,

This should be fixed in the newest commit:
https://github.com/rockchip-linux/gstreamer-rockchip/commit/75e767161c8a460045224994902674e308786692

So the mpp reset could not clear all resources(created with mpp_init), let's just recreate mpp every time.

whoman321, 2020-08-21:
This only clears 8 dma_buf objects, however there is 29 dma_buf objects every time the decoder is started.

Still not releasing 21 dma_buf objects with each start.  Still crashes after 20+ start and stop
So now, it is going (per each PLAYING state):

29 objects
51 objects
73 objects
95 objects
etc

JeffyCN, 2020-08-21:
cannot repro with your script, please try to update to newest mpp+gstreamer-rockchip+kernel, and also use fakesink to test.

JeffyCN, 2020-08-21:
well, i do repro with your original script(i was testing with xvimagesink/fakesink), so it seems like a knowning leaking in the old(1.10) kmssink, which likely to be fixed in:
https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad/-/commit/cfadd5a936d00312998c769ee48e64970692835a

whoman321, 2020-08-21:
Thanks for your extra testing.

in my original script, if I change from kmssink to rkximagesink, I have the same problem:
29 objects
51 objects
73 objects
95 objects
etc

whoman321, 2020-08-21:
Tomorrow I will compile the newest gst/plugins/xvimagesink and kmssink

I will let you know the results

Thank you very much for your help with this.

JeffyCN, 2020-08-21:
rkximagesink is based on kmssink ;)

anyway, please try fakesink, if still repro, please try update all packages(mpp/gst mpp/kernel)

closed 2020-08-24 by JeffyCN

whoman321, 2020-08-24:
You are correct.

I compiled gst-plugins-bad branch 1.10.4 with the kms patch, and all dma_buf refs are being cleared properly 

mppvideodec working properly after your fix
kmssink and rkximagesink working properly (newest versions) 

Thanks so much for everything

----------------------
unrelated note:  
in my example pipeline there is still a very very small continuous memory leak while the pipeline is running, and it jumps bigger when changing source.
I will do some testing later this week to see which elements are causing it.