rockchip-linux/gstreamer-rockchip #70: mppvideodec: very small continuous memory leak

whoman321, 2020-08-28:
When playing a video file using mppvideodec, the pipeline's memory usage increases very slowly, and is never released.

When changing the source, the memory leak rate jumps higher, then back to the slow leak while the video is playing.  Never released.

note:  this leak happens with both kmssink and fakesink

example3.py with mppvideodec:  memory leak
example3.py with avdec_h264: no memory leak

To show this, I modified my example3.py python script to automatically change source every 200ms. (line 124)

This accelerates the leak so that it is easily observable.

I've also tested the script with 60 seconds between source change, which shows the very slow continuous leak during playback.

I'm not sure what to do to find where within mppvideodec this leak is occuring.

Any ideas how we can proceed?

Thanks

```
import gc
import tty
import sys
import termios
import gi
import subprocess
import os, signal
import time
from time import sleep
gi.require_version('Gst', '1.0')
gi.require_version('GLib', '2.0')
gi.require_version('GObject', '2.0')
from gi.repository import Gst, GObject, GLib
GObject.threads_init()
Gst.init(sys.argv)

#GST PIPELINE
#GST - create the elements
source = Gst.ElementFactory.make("filesrc", "source")
demux = Gst.ElementFactory.make("qtdemux", "demux")
parse = Gst.ElementFactory.make("h264parse", "parse")
decoder = Gst.ElementFactory.make("mppvideodec", "decoder")
video_convert = Gst.ElementFactory.make("videoconvert", "videoconvert")
sink = Gst.ElementFactory.make("kmssink", "sink")

#GST - create the empty pipeline
p1 = Gst.Pipeline.new("test-pipeline")

if not p1 or not source or not sink:
	print("ERROR: Not all elements could be created")
	sys.exit(1)

#GST - add and link elements in pipeline
p1.add(source)
p1.add(demux)
p1.add(parse)
p1.add(decoder)
p1.add(video_convert)
p1.add(sink)

if not source.link(demux):
	print("ERROR: Could not link source to demux")
	sys.exit(1)
#if not demux.link(parse):
#	print("ERROR: Could not link demux to parse")
#		sys.exit(1)
if not parse.link(decoder):
	print("ERROR: Could not link parse to decoder")
	sys.exit(1)
if not decoder.link(video_convert):
	print("ERROR: Could not link decoder to video_convert")
#		sys.exit(1)
#if not decoder.link(sink):
if not video_convert.link(sink):
	print("ERROR: Could not link scaler to sink")
	sys.exit(1)

# set the source element's location property
source.set_property("location", "/video/test2.mp4")

ret = p1.get_state(1)
print ret.state

#GST - handler for the pad-added signal
def demux_on_pad_added(src, new_pad):
	if new_pad.get_name() == "video_0":
		print(
			"Received new pad '{0:s}' from '{1:s}'".format(
				new_pad.get_name(),
				src.get_name()))

		# attempt the link
		sink_pad = parse.get_static_pad("sink")
		new_pad_caps = new_pad.get_current_caps()
		new_pad_struct = new_pad_caps.get_structure(0)
		new_pad_type = new_pad_struct.get_name()
		print(new_pad_type)
		ret = new_pad.link(sink_pad)
		if not ret == Gst.PadLinkReturn.OK:
			print("Type is '{0:s}' but link failed".format(new_pad_type))
		else:
			print("Link succeeded (type '{0:s}')".format(new_pad_type))
		return
	return

# connect to the pad-added signal
demux.connect("pad-added", demux_on_pad_added)

p1.set_state(Gst.State.PLAYING)

#GST - Puase playback, set "location", and begin playback
def PlayVideo1():
#	GST stop pipeline
        p1.set_state(Gst.State.READY)

#	GST set filesrc 'location'
        source.set_property('location', '/video/test1.mp4')

#	GST start pipeline
        p1.set_state(Gst.State.PLAYING)
	
def PlayVideo2():
#	GST stop pipeline
        p1.set_state(Gst.State.READY)

#	GST set filesrc 'location'
        source.set_property('location', '/video/test2.mp4')
	
#	GST start pipeline
        p1.set_state(Gst.State.PLAYING)

for x in range(1, 5000, 1):
        print(x) 
        PlayVideo1()
        time.sleep(0.2)
        gc.collect()
        PlayVideo2()
        time.sleep(0.2)
        gc.collect()                    

#KEYBOARD - detect key press, and launch GST PlayVideo functions
orig_settings = termios.tcgetattr(sys.stdin)

tty.setcbreak(sys.stdin)
x = 0
while x != chr(27): # ESC
    x=sys.stdin.read(1)[0]
    print("You pressed", x)
    if x == "1":
        PlayVideo1()
    if x == "2":
        PlayVideo2()
    if x == "x":
        exit()
		
termios.tcsetattr(sys.stdin, termios.TCSADRAIN, orig_settings)
```

JeffyCN, 2020-09-01:
guessing you just cannot do it this way, since the state changing would create and post an async message to the pipleline bus, which needs to be handled...
1/ you need to add a bus watch to handle messages(especially the state changed messages in your case)
2/ you need to run the glib main loop

closed 2020-09-01 by JeffyCN

whoman321, 2020-09-03:
Hello,
I was incorrect.
There is no memory leak from continuous playback.

I made a mistake in my method for debugging memory usage, which caused me to think there was a leak.

I'm sorry to take your time on this.
Thanks again for all your help

JeffyCN, 2020-09-03:
You are welcome, i was using valgrind to trace it, which directly pointed to the async state changed messages, so it didn't take much time ;)

whoman321, 2020-09-04:
Could you show the full valgrind command line which you used to trace this?
Just asking for my own knowledge

JeffyCN, 2020-09-04:
> Could you show the full valgrind command line which you used to trace this?
> Just asking for my own knowledge

valgrind --show-leak-kinds=all --leak-check=full --log-file=/tmp/trace.log <cmd>

needs to install related dbgsym packages for symbols.