rockchip-linux/gstreamer-rockchip #72: About rk3308 videosink

Yann-Song, 2020-11-14:
Hi:
     which video sink is used for rk3308 ?  Here is my log

JeffyCN, 2020-11-20:
There are multiple video sink could be used, but that depends on which display service you are using.
Normally the rockchip buildroot SDK(with wayland display service) should support:
waylandsink kmssink glimagesink fakesink filesink fpsdisplaysink, .etc

For debian X11 sdk:
xvimagesink rkximagesink kmssink glimagesink ximagesink fakesink filesink fpsdisplaysink, .etc

Anyway, that's more about the system environment, and i don't think it has anything to do with gstreamer-rockchip....

closed 2020-11-20 by JeffyCN

Yann-Song, 2020-11-20:
Thank you !
    With use wayland or kmssink will cause oom, since rk3308 only  64M ddr.

JeffyCN, 2020-11-20:
well, that things indeed require lots of memory(dma) :(

maybe you can try fbdevsink, or write your own appsink and output to /dev/fb0

Yann-Song, 2020-11-20:
Ok , I will try it , thank you