rockchip-linux/gstreamer-rockchip #73: Invalid frame when encoding vp8

anders-holm-nextfour, 2020-11-17:
Hi,
I am trying to get vp8 encoding working  on a custom board using the mppvp8enc encoder and the following pipeline:

`gst-launch-1.0 videotestsrc pattern=ball ! \video/x-raw,format=NV12,width=960,height=540! mppvp8enc  ! rtpvp8pay ! udpsink host=10.10.10.110 port=5200 sync=false`

but  rtpvp8pay is not happy with the produced frame. A check in gstrvp8pay.c (around line 230) fails

    if (keyframe) {
        /* check start tag: 0x9d 0x01 0x2a */
        if (!gst_bit_reader_get_bits_uint8 (&reader, &tmp8, 8) || tmp8 != 0x9d) 
           goto error;

I am using:
 gstreamer 1.18
 gstreamer1-rockchip, commit 70c8a002a542412c6f5c44694ebcca62fc1c3d10
 mpp, commit  b0d0a5aafd328078d19455f606ddfd0f8103d7de

A similar  pipeline with h264 works ok:

`gst-launch-1.0 videotestsrc  pattern=ball ! \video/x-raw,format=NV12,width=960,height=540! mpph264enc ! rtph264pay ! udpsink host=10.10.10.110 port=5200 sync=false`

also the mpi_enc_test from mpp seems to work ok

    mpi_enc_test -t 9 -w 640 -h 272 -i /var/ds_480x272.yuv -o t.vp8 -f 4
    mpp[49085]: mpi_enc_utils: cmd parse result:
    mpp[49085]: mpi_enc_utils: input  file name: /var/ds_480x272.yuv
    mpp[49085]: mpi_enc_utils: output file name: t.vp8
    mpp[49085]: mpi_enc_utils: width      : 640
    mpp[49085]: mpi_enc_utils: height     : 272
    mpp[49085]: mpi_enc_utils: format     : 4
    mpp[49085]: mpi_enc_utils: type       : 9
    mpp[49085]: mpi_enc_test: mpi_enc_test start
    ....
    mpp[49085]: mpi_enc_test: 0x313f1b00 encoded frame 72   size 8131   
    mpp[49085]: mpi_enc_test: 0x313f1b00 encoded frame 73   size 8208   
   mpp[49085]: mpi_enc_test: 0x313f1b00 encoded frame 74   size 5723   
   mpp[49085]: mpi_enc_test: 0x313f1b00 found last frame. feof 1
   mpp[49085]: mpi_enc_test: 0x313f1b00 encoded frame 75   size 0      
   mpp[49085]: mpi_enc_test: 0x313f1b00 found last packet
   mpp[49085]: mpi_enc_test: 0x313f1b00 mpi_enc_test success total frame 76 bps 3356340

Any ideas about what could be wrong?

anders-holm-nextfour, 2020-11-17:
The problem seems to be that the mppvp8enc encoder outputs the frame in IVF format.

anders-holm-nextfour, 2020-11-17:
I added setting of the disable_ivf  configuration option in gst/rockchipmpp/gstmppvp8enc.c:gst_mpp_vp8_enc_update_properties

    mpp_enc_cfg_set_s32 (cfg, "vp8:qp_init", self->qp_init);
    mpp_enc_cfg_set_s32 (cfg, "vp8:qp_max", self->qp_max);
    mpp_enc_cfg_set_s32 (cfg, "vp8:qp_min", self->qp_min);
    +mpp_enc_cfg_set_s32 (cfg, "vp8:disable_ivf", 1);

and now the encoder outputs frames that rtpvp8pay is happy with.

JeffyCN, 2020-11-20:
Well, nice catch! I did noticed it and ask guys to remove the headers in MPP when adding this enc, but guessing i forgot to follow the process ;)

Will fix it soon, thanks.

closed 2020-11-20 by JeffyCN