rockchip-linux/gstreamer-rockchip #74: VP8 decoding fails with 'hal_vp8d_vdpu2_start send cmd failed'

anders-holm-nextfour, 2020-11-25:
Hi,

I am trying to use mppvideodec to decode VP8 using the following pipeline

`gst-launch-1.0 -v videotestsrc  ! "video/x-raw,width=640,height=480"  ! queue ! vp8enc ! mppvideodec ! filesink location=/tmp/test.out`

This fails with:
```
mpp[2125]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[2125]: hal_vp8d_vdpu2: hal_vp8d_vdpu2_start send cmd failed 12
```
which is caused by the kernel not being able to use an ion allocated buffer
```
[ 5944.200046] mpp_dma_import_fd:196: dma_buf_get fd 22 failed
[ 5944.206802] mpp_task_attach_fd:1236: can't import dma-buf 22
[ 5944.213596] mpp_translate_reg_address:1287: reg[ 63]: 0x00000016 fd 22 failed
```
Because I was running the pipeline with the ion_debug environment variable set to 0xff I can see that a buffer was allocated using the ion allocator, mmaped and then freed before it is used.
```
mpp[2125]: mpp_ion: allocator_ion_alloc enter: ctx 0x7f9804af00 size 614400
mpp[2125]: mpp_ion: ion_alloc enter: fd 20 len 614400 align 4096 heap_mask 1 flags 0
mpp[2125]: mpp_ion: ion_alloc leave: ret 0
mpp[2125]: mpp_ion: allocator_ion_alloc leave: ret 0 fd 22
...
mpp[2125]: mpp_ion: allocator_ion_mmap enter: ctx 0x7f9804af00 fd 22 size 614400
mpp[2125]: mpp_ion: allocator_ion_mmap leave: ret 0 ptr 0x7f95994000
...
mpp[2125]: mpp_ion: allocator_ion_free enter: ctx 0x7f9804af00 fd 22 ptr 0x7f95994000 size 614400
mpp[2125]: mpp_ion: allocator_ion_free leave
...
mpp[2125]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[2125]: hal_vp8d_vdpu2: hal_vp8d_vdpu2_start send cmd failed 12
```
The allocation, mmap and free happens in the gst_mpp_allocator_drm_buf function (gstmppallocator.c:256). The line causing the free is

`    mpp_buffer_group_put (group);`

If I remove this line the decoder works. I am not sure if this is an appropriate fix.

I am using a 5.4 kernel so I have had to modify the ion allocator in the mpp
library since the ion kernel interface has changed, but I do not think this is
caused by my changes since it is at an higher level and the call chain down to allocator_ion_free
is unchanged. Using the buffer after free seems wrong anyway.

mpp version is 3c5dc7fe4517f57dfb2619a59460a3e0ffc584fd
gstreamer-rockchip version is  b317184e9fb5bd7357afef274e500debe96676c5

JeffyCN, 2020-11-26:
seems like something wrong in the kernel's ion driver or mpp ion allocator.
the basic flow of the gst allocator is:
1/ alloc dma buffers from mpp drm/ion allocator
2/ import those buffers to mpp's external dma allocator
3/ destroy the original drm/ion allocator

so guessing the import part(allocator_ion_import) failed.

btw, we are using drm allocator in linux sdk, not ion. and if you are interesting on using ion, please consider tracing the import flow of the ion driver and compare with 4.4/4.19 bsp kernel

closed 2020-12-24 by JeffyCN

anders-holm-nextfour, 2021-03-17:
A late comment. Using the DRM allocator works. Just make sure that the correct DRM device is used.