rockchip-linux/gstreamer-rockchip #75: Latest source code cannot BUILD

thanh311078, 2020-12-10:
git clone https://github.com/rockchip-linux/gstreamer-rockchip.git
./autogen.sh
./configure
make

ERRORS:
gstmppvideoenc.h:80:3: error: unknown type name ‘MppEncHeaderMode’
gstmppvideoenc.c:97:7: error: unknown type name ‘MppEncHeaderMode’; did you mean ‘MppEncSeiMode’?
gstmppvideoenc.c:232:3: error: unknown type name ‘MppEncCfg’; did you mean ‘MppEncHwCfg’?
...
I try to search in source code and not see the MppEncHeaderMode, MppEncCfg... definition, may be missing the file?

thanh311078, 2020-12-10:
Sorry, need lastest mpp :-))

closed 2020-12-10 by thanh311078

coflery, 2021-04-20:
> 抱歉，需要最新的MPP :-)）
good job.
Me too.

willosonico, 2021-12-16:
how did you solve this? i still get the error

fghoussen, 2021-12-24:
Same problem. New to rockchip : what is mpp? How to update it?

fghoussen, 2021-12-24:
Will give a try to https://github.com/rockchip-linux/mpp