rockchip-linux/gstreamer-rockchip #77: rkximage on Ubuntu 20.04 (pipeline errors)

avafinger, 2021-01-26:
Compiled the latest gstreamer-rockchip in Ubuntu 20.04 but it gives me some weird errors.

**Gstreamer version:**
```
gst-launch-1.0 version 1.16.2
GStreamer 1.16.2
https://launchpad.net/distros/ubuntu/+source/gstreamer1.0

```
_**Plugins**_

**rockchipmpp:**
```
Plugin Details:
  Name                     rockchipmpp
  Description              Rockchip Mpp Video Plugin
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so
  Version                  1.10.0
  License                  LGPL
  Source module            gstreamer-rockchip
  Binary package           GStreamer Plug-ins source release
  Origin URL               Unknown package origin

  mppvideodec: Rockchip's MPP video decoder
  mpph264enc: Rockchip Mpp H264 Encoder
  mppvp8enc: Rockchip Mpp VP8 Encoder
  mppjpegenc: Rockchip Mpp JPEG Encoder
  mppjpegdec: Rockchip's MPP JPEG image decoder

  5 features:
  +-- 5 elements
```

**rkximage**
```
Plugin Details:
  Name                     rkximage
  Description              Rockchip X/DRM Video Sink
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrkximage.so
  Version                  1.10.0
  License                  LGPL
  Source module            gstreamer-rockchip
  Binary package           GStreamer Plug-ins source release
  Origin URL               Unknown package origin

  rkximagesink: Video sink

  1 features:
  +-- 1 elements

```

**rkisp:**
```
Factory Details:
  Rank                     none (0)
  Long-name                Gstreamer Plugin For Rockchip ISP Source
  Klass                    Source/Base
  Description              Capture camera video with rockchip ISP support
  Author                   Jacob Chen <cc@rock-chips.com>

Plugin Details:
  Name                     rkisp
  Description              rkisp
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrkisp.so
  Version                  1.0
  License                  unknown
  Source module            rkisp
  Binary package           librkisp
  Origin URL               https://github.com/01org/libxcam
[... 120 lines cut ...]
WARNING debug information: gstbasesink.c(3003): gst_base_sink_is_too_late (): /GstPlayBin:playbin/GstPlaySink:playsink/GstBin:vbin/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink:
There may be a timestamping problem, or this computer is too slow.

```
**Streaming from a camera:**
`gst-launch-1.0 rkisp device=/dev/video1 io-mode=4 ! 'video/x-raw,format=NV12,width=1280,height=720,framerate=30/1' ! rkximagesink
`
Error:
```
(gst-launch-1.0:19772): GStreamer-WARNING **: 18:17:59.639: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstcoreelements.so': /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstcoreelements.so: undefined symbol: gst_queue_array_peek_tail_struct
WARNING: erroneous pipeline: could not link xcamsrc0 to rkximagesink0 with caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720, framerate=(fraction)30/1

```

Saving to a file (rkisp issue?):
`gst-launch-1.0 rkisp device=/dev/video1 io-mode=4 ! 'video/x-raw,format=NV12,width=1280,height=720,framerate=30/1' ! mpph264enc ! filesink location=test.mp4 -ve`
Error:
```
(gst-launch-1.0:19734): GStreamer-WARNING **: 18:15:59.458: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstcoreelements.so': /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstcoreelements.so: undefined symbol: gst_queue_array_peek_tail_struct
WARNING: erroneous pipeline: no element "filesink"

```

What I am missing? Can you pinpoint the errors?
**rkisp1** works as it should, the gst-plugin rkisp and rkximage seems not working at all.

JeffyCN, 2021-01-27:
1/ rkisp&rkisp1 plugins are no longer using(not maintained), the BSP SDK is using rk(isp|aiq)-camera-engines(remote 3A daemon) + official v4l2src.

2/ the error log might means rkximagesink failed in "gst_x_image_sink_start()" or "gst_x_image_sink_xcontext_get()", you can add logs there to debug. this plugin request access to drm kernel driver and X display.

JeffyCN, 2021-01-27:
well, the isp error likely related to kernel driver, maybe open an issue there.

for rkximagesink, that log looks fine to me.
set window handle is only for overlay case(for example embeded in gui window like qt)
 if it still gives a black screen:

1/ check the result of drm set plane and page flip

2/ upload the output of:
 modetest -M rockchip
 cat /sys/kernel/debug/dri/0/summary

could be something wrong in the planes' zpos or displaying to the wrong monitor(rkximagesink only work with single monitor)

avafinger, 2021-01-28:
> (rkximagesink only work with single monitor)

Hmmm. That could be the case, for some weird reason, ubuntu created two displays, one virtual display and a second one, the physical display despite I have only one display. This happens in 4.19.161 with any desktop distro but not with FB only (without desktop). Unfortunately, there is no kmssink as you had in the extra that i could test.. 

I will investigate 1/ and 2/ and report back.

Thanks

JeffyCN, 2021-01-28:
you can try to use the original kmssink in gst plugin bad(needs libkms from libdrm)

and like kmssink, rkximagesink supports setting connector-id&plane-id property

i think caesar has ported a ubuntu system recently @Caesar-github

avafinger, 2021-01-29:
> and like kmssink, rkximagesink supports setting connector-id&plane-id property

You have a pipeline  sample handy to share?

> rkximagesink only work with single monitor

Soon as I removed the virtual display (which was the main display), rkximagesink worked.

Thank you!

closed 2021-01-29 by avafinger

avafinger, 2021-01-31:
Hi @JeffyCN 

A question for you.

I am now with Kernel 4.19.161 (Rockchip) and compared the Images taken with gstreamer (on Kernel 4.19.161) and with FriendlyElec Image (on Kernel 4.4.179). The sensor is ov4689.

The new rkisp engine grab frames with noise (and darker) while in FE not,  can you advise how to fix it or should I open an issue in camera-engine? I cannot use their iqfile, it throws errors during the parsing. 

Can you advise?

**FE grabbed Image (Kernel 4.4.179)**
[image]

**New rkisp grabbed Image (Kernel 4.19.161)**
[image]

JeffyCN, 2021-01-31:
could be version mismatched, you can try the newest kernel 4.19 on my github's mirrors repo.
or maybe send a mail to ask zhengsq@rock-chips.com