rockchip-linux/gstreamer-rockchip #79: qt gstreamer custom pipeline

kadirbeytorun, 2021-03-03:
Hello,
I been trying to construct custom pipeline in qmultimedia class Qmediaplayer, using your sdk as reference.

So far, I couldn't figure out which sink you use in your implementations. I know I can use waylandsink directly for gstreamer, but for Qmediaplayer, I dont know which sink to use.

From Qt's documentation, it shows that they use xvimagesink
QMediaPlayer *receiver = new QMediaPlayer ;
videoWidget = new QVideoWidget ; receiver->setVideoOutput(videoWidget); receiver->setMedia( QUrl ("gst-pipeline: udpsrc port=5000 ! \     application/x-rtp,encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! \     xvimagesink name=qtvideosink")); receiver->play(); // Content will be shown in this widget. videoWidget->show();

But xvimagesink is not even compiled as default, so rockchip-gst cannot be using it.
Could you explain which sink rockchip uses for qtmultimedia? And how can I integrate it to qmediaplayer to create custom gst pipelines in Qt

JeffyCN, 2021-03-03:
ximagesink depends on x11. 

on our sdk, x11 use rkximagesink/xvimagesink
wayland use kmssink/waylandsink

our buildroot sdk use environment to specify sink for qtmultimedia.(grep -r SINK /etc/)

closed 2021-03-03 by JeffyCN

kadirbeytorun, 2021-03-03:
I know that qt uses waylandsink when "QT_GSTREAMER_WINDOW_VIDEOSINK=waylandsink" environment parameter is set.

My question is about setting custom pipelines in qmediaplayer. (if I put waylandsink in the pipeline, it doesnt work as intended) 
I tried using xvimagesink (as instructed in QT's documents), but got other problems regarding x11

How can I construct custom gst pipeline specifically in Qmediaplayer?

 receiver->setMedia( QUrl ("gst-pipeline: udpsrc port=5000 ! \ application/x-rtp,encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! \ xvimagesink name=qtvideosink")) Such as this, but without xvimagesink.

JeffyCN, 2021-03-03:
i don't think this has anything to do with gst-rockchip, it's pure gst + qtmultimedia.

maybe try to ask it in qt mail list?

and that feature should be available since qt 5.12.2

kadirbeytorun, 2021-03-03:
Yeah I am on working on 5.12 version, so it should not be a problem.

Its not pure gst and qt problem, because I can use custom pipelines in desktop environment(probably because its already using x11 as default display manager)

My question is about how to make it work with waylandsink. I observed that there were some patches you added in buildroot to add support for qt5multimedia to use waylandsink. I am not really sure how it works exactly, you used waylandsink and yet at the same time Qt has control over the frames.
https://github.com/rockchip-linux/buildroot/tree/rockchip/2018.02-rc3/package/qt5/qt5multimedia/5.12.2

If I was using default cases for qmediaplayer and qcamera, qt will automatically handle the sinks and I dont need to worry about anything. But when custom pipeline is the deal, I have to provide which sink the video needs to flow.
Do I need to make changes in qt5multimedia source codes to handle this? Or make some changes in my pipeline (especially regarding which sink to use)?

JeffyCN, 2021-03-03:
first of all, please notice, it's completed pure gst+qt issue.

desktop uses x11, which is *not* the same as our buildroot, qtmultimedia supports x11 officially by default for a long time, but lake of supports for waylandsink.

so it's much about:
1/ confirm custom pipeline works in qtmultimedia <--- qt issue
2/ support waylandsink in qtmultimedia's custom pipeline flow <--- qt + gst issue

anyway, i don't think this issue belongs to here(gst-rockchip plugin)...

JeffyCN, 2021-03-03:
for wayland, it's also possible to use kmssink(directly render to DRM) and maybe glimagesink with:
export GST_GL_API=gles2
export GST_GL_PLATFORM=egl

JeffyCN, 2021-03-03:
please open issue in buildroot repo, and maybe try the newest version which is using qt5.14

JeffyCN, 2021-03-04:
FYI, that works well on my rk3399 evb(with newest buildroot sdk):

int main(int argc, char *argv[])
{
    QApplication app(argc, argv);

    QVideoWidget w;

    w.setGeometry(0, 0, 1024, 800);
    w.show();

    QMediaPlayer *player;

    player = new QMediaPlayer;
    player->setVideoOutput(&w);
    player->setMedia(QMediaContent(QUrl(argv[1])));
    player->play();

    return app.exec();
}

videowidgettest "gst-pipeline: videotestsrc ! waylandsink name=qtvideosink"
videowidgettest "gst-pipeline: videotestsrc ! kmssink name=qtvideosink"
videowidgettest "gst-pipeline: v4l2src ! waylandsink name=qtvideosink"
videowidgettest "gst-pipeline: v4l2src ! kmssink name=qtvideosink"

kadirbeytorun, 2021-03-04:
Well I am on rk1808-evb, and cannot really use the latest version that I downloaded from
https://github.com/rockchip-linux/manifests
Because it wont build at default configuration
I opened issue there about it
https://github.com/rockchip-linux/manifests/issues/17
No one bothered to reply

I will try fixing the errors for new version of sdk building nonetheless and see where it leads me, thank you for your great support.

Also I didnt know we can give qt waylandsink as such, "waylandsink name=qtvideosink", in my case when I type waylandsink in custom pipeline, it will just output the video directly to drm and not in QVideoWidget

JeffyCN, 2021-03-04:
that manifest is not maintained for a long time(for some bussiness reason).

newer sdk should be provided by the customer services(from rockchip chinese git server).

the related buildroot packages would be weston/gstreamer1(plugin bad)/qt5/wayland*, there might be some related patches recently.

kadirbeytorun, 2021-03-04:
I do not have access to rockchip chinese git server unfortunately (redmine rockchip server). Is there no other way for me to access latest version of sdk? Perhaps you can share through onedrive or baidudisk? Or perhaps share updated version of manifest file of rk1808.

I couldn't see anything like "weston/gstreamer1(plugin bad)/qt5/wayland*" in buildroot patches. Nonetheless, it appears all my problems are from using old version of sdk.

JeffyCN, 2021-03-04:
please contact service@rock-chips.com for access permission, i cannot provide it since the manifest point to private git server too(not all in this github, and require access permission).

the related patches should be in:
https://github.com/JeffyCN/rockchip_mirrors/tree/buildroot/package/qt5
https://github.com/JeffyCN/rockchip_mirrors/tree/buildroot/package/gstreamer1/gst1-plugins-bad/

the newest qt+waylandsink fixes are landed in 2021 Jan 19

kadirbeytorun, 2021-03-05:
Thanks for your time and great support @JeffyCN 

I contacted service@rock-chips but did not get any reply so far.

I will examine those patches and try to make them work in my old version sdk. (Dont know how though)

In the meantime, would it be possible for you guys to update those outdated rk1808 manifests? Or perhaps help me get access to redmine rockchip repositories. (I registered months ago and didnt get approved, my username is "rakidis")

JeffyCN, 2021-03-08:
sorry, i cannot provide new manifest, since it pointed to the *private* git server, which requires ssh keys. the only way to access it would be contact the customer service window and add your ssh key to it(guessing you can ask them for redmine account too).

i've built a 1808 image(without npu stuff) for testing qt + gst custom pipleline:
https://drive.google.com/file/d/1QhpJKTRMr4C-9OYr37ZQEuTQQpy9sd5L/view?usp=sharing