rockchip-linux/gstreamer-rockchip #82: mpph264enc IDR (key frame) request

rw-vemotion, 2021-07-28:
I'm adding Rockchip support to an existing application, but it appears that requests for an IDR, using either _gst_video_event_new_upstream_force_key_unit_ or _gst_video_event_new_downstream_force_key_unit_ events do not result in an IDR being generated.

As an experiment, I adjusted the GOP to 500 (20 seconds @ 25 fps) and no IDR was generated until after 20 seconds. I would have expected an IDR immediately following the request.

JeffyCN, 2021-07-29:
that is not supported yet, but if you can provide a way to test(demo or command), maybe i can try to add it.

the mpp has a MPP_ENC_SET_IDR_FRAME cmd for that, but only for h264&h265, but i've done something similar with gop(1):
https://github.com/JeffyCN/libv4l-rkmpp/commit/bf9ea3d04a2a3831867e903993fca7c9a60bb1fb

JeffyCN, 2021-07-29:
try this:
https://github.com/JeffyCN/rockchip_mirrors/commit/0a386aeb4ceca2bf6551a1431f0492af0d14f9bd

rw-vemotion, 2021-08-03:
Well I've finally managed to manually apply the changes (for the second patch only) but this has not made any difference to the behaviour.
After starting stream (with GOP=500) it still takes 20 seconds before it commences playing.
Switching between VBR and CBR whilst running results in image corruption (pink splodges) for 20 seconds before the key frame is received

JeffyCN, 2021-08-03:
Why not just use the newest plugin?
Could not repro this issue.

a simple test:
https://paste.ubuntu.com/p/Q5vSfxYxj9/

gst-launch-1.0 videotestsrc ! mpph264enc ! fakesink 2>/dev/null
gst_mpp_enc_loop[786] got keyframe: 0
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 0
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 0
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 0
gst_mpp_enc_loop[764] request keyframe: 1
gst_mpp_enc_loop[786] got keyframe: 1
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 1
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 0
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 0
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 0
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 0
gst_mpp_enc_loop[764] request keyframe: 0
gst_mpp_enc_loop[786] got keyframe: 0

closed 2021-08-03 by JeffyCN

JeffyCN, 2021-08-03:
BTW, i'm using the newest MPP:
https://github.com/JeffyCN/rockchip_mirrors/commit/b212aa4d08b106c2d3ce1b4da5cde5cd80a5bc31

There also logs when gop changed(every change would cause keyframe):
mpp[1015]: mpp_enc: MPP_ENC_SET_RC_CFG bps 288000 [270000 : 306000] fps [30:30] gop 1
mpp[1015]: mpp_enc: MPP_ENC_SET_RC_CFG bps 288000 [270000 : 306000] fps [30:30] gop 30

https://github.com/JeffyCN/rockchip_mirrors/blob/mpp/mpp/codec/mpp_enc_impl.cpp#L310
https://github.com/JeffyCN/rockchip_mirrors/blob/mpp/mpp/codec/mpp_enc_impl.cpp#L849

rw-vemotion, 2021-08-03:
Unfortunately, I'm building under Firefly-Linux and they are not in sync with Rockchip.
When you first posted, I couldn't even apply your fix as the repo were so out of sync that it didn't have the filename change.

JeffyCN, 2021-08-03:
Okay then, this change might not be available for old repo, it depends on the new enc property APIs and MPP.

Guess you can add logs to make sure the gop did changed when getting a keyframe request. And maybe keep the gop to 1 for a few frames untill getting a real keyframe(check the packet meta like the diff i post above)

rw-vemotion, 2021-08-03:
Yes. That's the plan. I've been hit by Ubuntu freezes today, so it's taking longer than planned.