rockchip-linux/gstreamer-rockchip #89: Question: videoscale with RGA acceleration

whoman321, 2021-12-16:
Hello Jeffy,
I hope you are well my friend!

What is the current method to use RGA for videoscale in gstreamer?  

For Pixel Format conversion, I'm using your gst-plugins-base code with RGA (videoconvert), and it is working perfectly.
https://github.com/JeffyCN/gst-plugins-base

But I cannot find how to use videoscale with RGA acceleration.

I know that rgaconvert is deprecated, and I cannot find the new way to to videoscale.

What should I use for videoscale? (RGA accelerated)

I'm very sorry if this is not the correct place to ask this question.

Thank you for your time.

JeffyCN, 2021-12-16:
hi, the newest mppvideodec should be able to set scaled size(use width and height properties) as well as output pixel format (uaing rga). the other way is using render-rectangle property of the sink element.

i can try to add rga support in videoscale too, but i would prefer above 2 ways

whoman321, 2021-12-16:
That's great!  Yes I would prefer to only use mppvideodec for conversion and scaling, so no need for videoscale.

I will clone now and test it out.
Thank you so much.  You're the man!

whoman321, 2021-12-16:
I made a mistake, I do need to use a separate element if possible.

Here is what I'm trying to do:
1 source to 2 sinks (different scale and format)

h264 -> mppvideodec -> tee name=t 
t. -> kmssink
t. -> videoconvert -> videoscale -> v4l2sink

I need to split the h264 and send to two sinks.  One sink with no conversion/scale, and the second sink with format conversion and scale.

I'm sorry for my mistake.

If there is any chance you can try to add rga support in videoscale too, that would be a major help for our product.

Thank you again for your time and help.

JeffyCN, 2021-12-16:
well, i can try it, but this seems slow: conversion + scaling, and using non-dma buf.

or if the size of 2 sinks are similar, maybe you can try to convert to sink 2's format and size in mppvideodec, then use kmssink's render-rectangle(vop scaling) in sink 1.

anyway, i'll try to check videoscale tomorror.

whoman321, 2021-12-16:
Thank you.  Hopefully it's possible.

Unfortunately I need the scale to be different because at the v42lsink, the size should be small (640x360), and this would be too small for kmssink (main display - full video resolution).

If it's possible with videoscale, that would be great.  If it's not possible, I can try to use the old rgaconvert?  I will try tomorrow.

Have a good night, bud!

JeffyCN, 2021-12-17:
well, seems like i've done that before, but didn't push to that repo...

check:
https://github.com/JeffyCN/rockchip_mirrors/blob/buildroot/package/gstreamer1/gst1-plugins-base/0044-video-converter-Support-rockchip-RGA-2D-accel.patch#L249

whoman321, 2021-12-23:
Thank you so much for the patch link.  You're the best!

After testing, RGA is not working for me, for videoconvert or videoscale.

I did insert this code for videoscale:
```{
  int i, n_planes;

+GST_WARNING ("*******************RGA TEST 1 *************************");
+#ifdef HAVE_RGA
+  GST_WARNING ("*******************RGA scale planes confirm *************************");
+  if (!GST_VIDEO_FRAME_IS_INTERLACED (src)) {
+    /* Accel convert with rockchip RGA */
+    if (video_converter_try_rga (convert))
+      return;
+  }
+#endif

  n_planes = GST_VIDEO_FRAME_N_PLANES (dest);
```

I now realize that before, I was not using RGA for videoconvert, because I did not have the environment variable set: GST_VIDEO_CONVERT_USE_RGA=1   

If I enable RGA on my gst pipeline, and use videoconvert or videoscale, I get a SIGSEGV every time.  With no GST_VIDEO_CONVERT_USE_RGA=1, it works, but drops many frames because of software conversion and scaling.

RGA enabled = SIGSEGV:
```mpp[15449]: mpp: deprecated block control, use timeout control instead
mpp[15449]: mpp: deprecated block control, use timeout control instead
0:00:01.573683718 15449   0x7f78007800 WARN         video-converter video-converter.c:4244:convert_scale_planes: *******************RGA TEST 1 *************************
0:00:01.573866591 15449   0x7f78007800 WARN         video-converter video-converter.c:4247:convert_scale_planes: *******************RGA scale planes confirm *************************
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 15449' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

Do I have a problem with my RGA setup?

```cat .config|grep RGA
CONFIG_VIDEO_ROCKCHIP_RGA=y
CONFIG_ROCKCHIP_RGA2=y
```

librga and librga-dev from: https://github.com/rockchip-linux/linux-rga

```ls -l /dev/rga
crw------- 1 root root 10, 57 Dec  9 00:01 /dev/rga
```

for gst-plugins-base I'm using: https://github.com/JeffyCN/gst-plugins-base/tree/1.10

I do not know what else to try.

Maybe I'll try recompiling with newest gstreamer-rockchip.
Currently I'm using gstreamer-rockchip that I compiled it 1 year ago (with ION patch) from: https://github.com/rockchip-linux/gstreamer-rockchip

Let me know if you have any ideas?

JeffyCN, 2021-12-23:
that patch is tested on 1.14, so guessing due to some api changes.

you can try to use gdb or add more logs to locate the crashed code(likely somewhere in video_converter_try_rga)

whoman321, 2021-12-28:
Hello,
Sorry for the delay (busy week).

To avoid api issues I started with a clean new rootfs...

I compiled Debian Buster from: https://github.com/rockchip-linux/rk-rootfs-build

I compiled gstreamer-rockchip from: https://github.com/JeffyCN/rockchip_mirrors/commit/31fb9e2d56444aeeb4a46c4bb54709deb3043abd

mppvideodec is working using hardware (RGA) scaling and pixel format conversion.

I compiled gst-plugins-base from: https://github.com/JeffyCN/gst-plugins-base/tree/1.14
I chose the 1.14 branch, since rockchip buster uses gstreamer 1.14 **(is this the correct branch?)**

**Testing video convert RGA:**

**_without RGA enabled:_**
```
gst-launch-1.0 filesrc location=/home/linaro/spiral.mp4 ! qtdemux ! h264parse ! mppvideodec width=640 height=360 ! videoconvert n-threads=4 ! video/x-raw,format=YUY2 ! tee ! v4l2sink device=/dev/video1
```
Result:  gst-launch-1.0 CPU usage is 140% to 170% (as expected since videoconvert is not using RGA)

**_with RGA enabled:_**
```
sudo GST_VIDEO_CONVERT_USE_RGA=1 gst-launch-1.0 filesrc location=/home/linaro/spiral.mp4 ! qtdemux ! h264parse ! mppvideodec width=640 height=360 ! videoconvert n-threads=4 ! video/x-raw,format=YUY2 ! tee ! v4l2sink device=/dev/video1
```
Result:  gst-launch-1.0 CPU usage is 140% to 170% (not as expected videoconvert should be using RGA but it's using SW)

It seems strange that mppvideodec is using RGA for scaling, however videoconvert is not using RGA with **GST_VIDEO_CONVERT_USE_RGA=1**

Is there a step that I have missed in order to use this gst-plugins-base 1.14 with RGA?
Am I using the correct branch (1.14)?

Before I try to patch/test videoscale, I just want to make sure that this gst-plugins-base (videoconvert) is working with RGA.  

I will try adding more logs to see if I can find where the problem might be.
I'll post my results from adding more logs later tonight or tomorrow.

If you have any thoughts, please let me know.

Thanks so much

JeffyCN, 2021-12-28:
https://github.com/JeffyCN/gst-plugins-base/tree/1.14 is out-dated for a long time.

the newest patches are within rockchip buildroot sdk:
https://github.com/JeffyCN/rockchip_mirrors/blob/buildroot/package/gstreamer1/gst1-plugins-base/0044-video-converter-Support-rockchip-RGA-2D-accel.patch

@jair.wu@rock-chips.com has ported those patches to debian sdk:
https://github.com/JeffyCN/rockchip_mirrors/commit/ed04efc66a7f385c29aafbfd1da848f48c80a7bd

whoman321, 2021-12-28:
Oh Great!  Thank you soooo much!
I installed gst-plugins-base debs and now videoconvert is working well with RGA  :)
Now I would like to patch/test the source code, to add RGA to videoscale.
Can you let me know which repository is the source code which I should apply the videoscale patch to?

JeffyCN, 2021-12-28:
they are in the same patch:
https://github.com/JeffyCN/rockchip_mirrors/blob/buildroot/package/gstreamer1/gst1-plugins-base/0044-video-converter-Support-rockchip-RGA-2D-accel.patch#L249

the deb should have this function as well

JeffyCN, 2021-12-28:
for more details about how to build those deb, please mail jair.wu@rock-chips.com (he might not have github account)

whoman321, 2021-12-28:
Oh I'm sorry...
You are correct.  I just tested and videoscale is working with RGA

Jeffy, you are the best!

For now this package is enough for me to continue work on our software, and if/when I may need the source, I will email Jair Wu.
Thank you again for all of your help.

closed 2021-12-28 by whoman321