rockchip-linux/gstreamer-rockchip #90: failed to build ‘GstDmaBufAllocator’ undeclared

willosonico, 2021-12-16:
trying to compile in asus tinker edge r, with debian stretch 9.12 and gstreamer-1.0
maybe i should use a specific commit compatible with my environment? if so, how can i find it?

In file included from /usr/lib/aarch64-linux-gnu/glib-2.0/include/glibconfig.h:9:0,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/gstreamer-1.0/gst/gst.h:27,
                 from /usr/include/gstreamer-1.0/gst/allocators/gstdmabuf.h:24,
                 from gstmppallocator.c:28:
gstmppallocator.c:34:20: error: ‘GstDmaBufAllocator’ undeclared (first use in this function)
     MPP_ALLOCATOR, GstDmaBufAllocator);

JeffyCN, 2021-12-17:
this is a new api in gst 1.12

the plugin is going to bump min gst version to 1.14 (debian10)

JeffyCN, 2021-12-17:
for this error only:
https://filebin.net/79zkbqdn964irimz/gst-rkmpp-1.0.diff

closed 2021-12-17 by JeffyCN

willosonico, 2021-12-17:
hi @JeffyCN, if a patch the repo i get first a patch error

linaro@linaro-alip:~/sdcard/gstreamer-rockchip$ patch -i diff
can't find file to patch at input line 5
Perhaps you should have used the -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/gst/rockchipmpp/gstmppallocator.c b/gst/rockchipmpp/gstmppallocator.c
|index 1a0224e..1cd8f75 100644
|--- a/gst/rockchipmpp/gstmppallocator.c
|+++ b/gst/rockchipmpp/gstmppallocator.c
--------------------------
File to patch: gst/rockchipmpp/gstmppallocator.c
patching file gst/rockchipmpp/gstmppallocator.c
Hunk #4 FAILED at 302.
1 out of 4 hunks FAILED -- saving rejects to file gst/rockchipmpp/gstmppallocator.c.rej

linaro@linaro-alip:~/sdcard/gstreamer-rockchip$ cat  gst/rockchipmpp/gstmppallocator.c.rej
--- gstmppallocator.c
+++ gstmppallocator.c
@@ -302,7 +300,7 @@ gst_mpp_allocator_init (GstMppAllocator * allocator)
 {
   GstAllocator *alloc = GST_ALLOCATOR_CAST (allocator);
 
-  alloc->mem_type = GST_ALLOCATOR_MPP;
+  alloc->mem_type = GST_ALLOCATOR_DMABUF;
   alloc->mem_map_full = GST_DEBUG_FUNCPTR (gst_mpp_mem_map_full);
 
   GST_OBJECT_FLAG_SET (allocator, GST_ALLOCATOR_FLAG_CUSTOM_ALLOC);

then after recompiling

gstmppallocator.c:34:20: error: ‘GstFdAllocator’ undeclared (first use in this function)
     MPP_ALLOCATOR, GstFdAllocator);

if i compile the https://github.com/TinkerBoard/gstreamer-rockchip i get this error instead

gstmppvideodec.c: In function ‘gst_mpp_video_dec_open’:
gstmppvideodec.c:188:18: error: ‘MPP_IO_MODE_SIMPLE’ undeclared (first use in this function)
   gint io_mode = MPP_IO_MODE_SIMPLE;
                  ^~~~~~~~~~~~~~~~~~
gstmppvideodec.c:188:18: note: each undeclared identifier is reported only once for each function it appears in
gstmppvideodec.c:192:38: error: ‘MPP_SET_TRANSACTION_MODE’ undeclared (first use in this function)
   self->mpi->control (self->mpp_ctx, MPP_SET_TRANSACTION_MODE,

my problem is in fact that mppvideodec seems not to be working on default installation of tinker edge r (debian 9)

mpp[27539]: mpp_rt: NOT found ion allocator
mpp[27539]: mpp_rt: found drm allocator
mpp[27539]: mpp: deprecated block control, use timeout control instead
mpp[27539]: h264d_api: is_avcC=1
Caught SIGSEGV
exec gdb failed: No such file or directory

the pipeline i used is 

gst-launch-1.0 rtspsrc location=rtsp://192.168.1.12:554/h264Preview_01_sub latency=0 \
! rtph264depay \
! h264parse \
! mppvideodec \
! videoconvert \
! fpsdisplaysink

i can actually use avdec_h264 instead of mppvideodec, but it's decoding on hardware? 

thank you

JeffyCN, 2021-12-17:
this repo is a little old, newest plugin eould be here:
https://github.com/JeffyCN/rockchip_mirrors/commit/31fb9e2d56444aeeb4a46c4bb54709deb3043abd

i do have a custom ffmpeg for hardware decoding, but that would require an extra frame conversion:
https://github.com/JeffyCN/FFmpeg

JeffyCN, 2021-12-17:
and this plugin is a wrapper of mpp, so it depends on mpp and librga and related kernel drivers.
 so it looks like the tinker version is using a custom mpp.

the newest official mpp is:
https://github.com/JeffyCN/rockchip_mirrors/tree/mpp

whoman321, 2021-12-17:
I'm experiencing the same issue.  Thank you for the patch, and the suggestion to include gstfdmemory.h

Still some problems...

Environment:
Debian Stretch
gstreamer from: libgstreamer1.0-dev 1.10.4-1 from debian mirror
Newest MPP from: https://github.com/rockchip-linux/mpp
gstreamer-rockchip from: https://github.com/JeffyCN/rockchip_mirrors/commit/31fb9e2d56444aeeb4a46c4bb54709deb3043abd

After the patch and include gstfdmemory.h, error:
```

./gst/rockchipmpp/gstmpp.c:57:5: error: ‘GST_VIDEO_FORMAT_VYUY’ undeclared here (not in a function)
   { GST_VIDEO_FORMAT_ ## gst, MPP_FMT_ ## mpp, RK_FORMAT_ ## rga, stride, yuv }
     ^
../gst/rockchipmpp/gstmpp.c:79:3: note: in expansion of macro ‘GST_MPP_FORMAT’
   GST_MPP_FORMAT (VYUY, YUV422_VYUY, UNKNOWN, 1, 1),
   ^~~~~~~~~~~~~~
../gst/rockchipmpp/gstmpp.c: In function ‘gst_mpp_video_info_align’:
../gst/rockchipmpp/gstmpp.c:324:3: error: invalid use of void expression
   if (!gst_video_info_align (info, &align))
   ^~
../gst/rockchipmpp/gstmpp.c:324:7: error: wrong type argument to unary exclamation mark
   if (!gst_video_info_align (info, &align))
       ^

```
I did try comment out line 79 in gstmpp.c just to test:
`
//  GST_MPP_FORMAT (VYUY, YUV422_VYUY, UNKNOWN, 1, 1),
`
and still more errors after that:
```

../gst/rockchipmpp/gstmpp.c: In function ‘gst_mpp_video_info_align’:
../gst/rockchipmpp/gstmpp.c:324:3: error: invalid use of void expression
   if (!gst_video_info_align (info, &align))
   ^~

```

Perhaps I should try a different commit?

whoman321, 2021-12-18:
Thank you, that worked.
next error:
```
gst/rockchipmpp/503b108@@gstrockchipmpp@sha/gstmpph264enc.c.o: In function `gst_mpp_h264_enc_set_src_caps':
./obj-aarch64-linux-gnu/../gst/rockchipmpp/gstmpph264enc.c:267: undefined reference to `g_enum_to_string'
./obj-aarch64-linux-gnu/../gst/rockchipmpp/gstmpph264enc.c:271: undefined reference to `g_enum_to_string'
```

JeffyCN, 2021-12-18:
it's a new api in newer glib(2.56 maybe)

you can try to hard-code profile to "high" and level to "40" (which are the default settings)