rockchip-linux/gstreamer-rockchip #91: mppvideodec hardware/sotware conversion to RGB

willosonico, 2021-12-29:
i'm using mppvideodec gstreamer plugin to decode an h264 stream, but the output is not in rgb, and if i use videoconvert the framerate drops because it runs on the cpu, there is a way to hardware convert mppvideodec output in rgb? thankyou

JeffyCN, 2021-12-30:
try: https://github.com/JeffyCN/rockchip_mirrors/commits/gstreamer-rockchip
export GST_MPP_VIDEODEC_DEFAULT_FORMAT=RGB

willosonico, 2021-12-30:
in ubuntu 18.04 from firefly vendor, i did

git clone -b gstreamer-rockchip --single-branch  https://github.com/JeffyCN/rockchip_mirrors
cd rockchip_mirrors 
meson setup build

....
Header <gst/video/video-format.h> has symbol "GST_VIDEO_FORMAT_NV12_10LE40" with dependency gstreamer-video-1.0: NO 
Configuring config.h using configuration
Program python3 found: YES (/usr/bin/python3.6)
Build targets in project: 2

Found ninja-1.8.2 at /usr/bin/ninja

then
cd build
ninja

...
../gst/rockchipmpp/gstmppvideodec.c:252:52: error: ‘MPP_FRAME_FBC_AFBC_V2’ undeclared (first use in this function); did you mean ‘MPP_FRAME_FBC_AFBC_V1’?
     MppFrameFormat mpp_format = MPP_FMT_YUV420SP | MPP_FRAME_FBC_AFBC_V2;
...

there's an instruction on how to build? am i missing some dependency? thank you

JeffyCN, 2021-12-30:
this plugin is a wrapper for mpp lib, so the newer version depends on newer mpp.

https://github.com/JeffyCN/rockchip_mirrors/commits/mpp

and the newer mpp might also depends on newer mpp kernel driver...

for this build error only, you can just remove this MPP_FRAME_FBC_AFBC_V2, or replace it to 0

closed 2022-01-10 by JeffyCN

willosonico, 2022-01-19:
tryed with removing MPP_FRAME_FBC_AFBC_V2 but i still got this error

[6/6] Linking target gst/rockchipmpp/libgstrockchipmpp.so
FAILED: gst/rockchipmpp/libgstrockchipmpp.so 
cc  -o gst/rockchipmpp/libgstrockchipmpp.so gst/rockchipmpp/libgstrockchipmpp.so.p/gstmpp.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppallocator.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppdec.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppjpegdec.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppvideodec.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppenc.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppjpegenc.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmpph264enc.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmpph265enc.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppvp8enc.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppalphadecodebin.c.o gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppvpxalphadecodebin.c.o -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,libgstrockchipmpp.so /usr/lib/aarch64-linux-gnu/libgstbase-1.0.so /usr/lib/aarch64-linux-gnu/libgstreamer-1.0.so /usr/lib/aarch64-linux-gnu/libgobject-2.0.so /usr/lib/aarch64-linux-gnu/libglib-2.0.so /usr/lib/aarch64-linux-gnu/libgstvideo-1.0.so /usr/lib/aarch64-linux-gnu/libgstallocators-1.0.so /usr/lib/aarch64-linux-gnu/libgstpbutils-1.0.so /usr/lib/aarch64-linux-gnu/libgstaudio-1.0.so /usr/lib/aarch64-linux-gnu/librockchip_mpp.so /usr/lib/aarch64-linux-gnu/librga.so -Wl,--end-group
gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppdec.c.o: In function `gst_mpp_dec_apply_info_change':
/home/firefly/JeffyCN/rockchip_mirrors/build/../gst/rockchipmpp/gstmppdec.c:423: undefined reference to `mpp_frame_get_offset_x'
/home/firefly/JeffyCN/rockchip_mirrors/build/../gst/rockchipmpp/gstmppdec.c:424: undefined reference to `mpp_frame_get_offset_y'
gst/rockchipmpp/libgstrockchipmpp.so.p/gstmppdec.c.o: In function `gst_mpp_dec_get_gst_buffer':
/home/firefly/JeffyCN/rockchip_mirrors/build/../gst/rockchipmpp/gstmppdec.c:663: undefined reference to `mpp_frame_get_offset_x'
/home/firefly/JeffyCN/rockchip_mirrors/build/../gst/rockchipmpp/gstmppdec.c:664: undefined reference to `mpp_frame_get_offset_y'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

JeffyCN, 2022-01-19:
there's also a patch to add hw accel to videoconvert:
https://github.com/JeffyCN/rockchip_mirrors/blob/buildroot/package/gstreamer1/gst1-plugins-base/1.14.4/0048-video-converter-Support-rockchip-RGA-2D-accel.patch