rockchip-linux/gstreamer-rockchip #93: MPEG-2 hardware decoding not smooth

LianYangCn, 2022-02-15:
Source: MPEG-2 720x576 25FPS 2500KB/s local file
Use https://github.com/rockchip-linux/gstreamer-rockchip/commit/ab044195bfb472423c3465bfc26f9fd67fe7383b

LianYangCn, 2022-02-15:
I know why the warnings displayed

Because the timestamps (PTS) of the frame are not monotonically increasing.
Monotonically increasing:
123 456 789 ...
The MPEG-2 file got:
312 645 987 ...

But how to fix it?

JeffyCN, 2022-02-15:
try the newest version in:
https://github.com/JeffyCN/rockchip_mirrors/tree/gstreamer-rockchip

LianYangCn, 2022-02-16:
> try the newest version in: https://github.com/JeffyCN/rockchip_mirrors/tree/gstreamer-rockchip
> 
> and provide the video file if still has issues.

Exceptions:

[image]

JeffyCN, 2022-02-16:
try newest mpp too:
https://github.com/JeffyCN/rockchip_mirrors/tree/mpp

LianYangCn, 2022-02-16:
Test MPG
[mpg.zip](https://github.com/rockchip-linux/gstreamer-rockchip/files/8075938/mpg.zip)

LianYangCn, 2022-02-16:
yo ho, same error
latest MPP with gstreamer-rockchip

```log
Setting pipeline to PAUSED ...
0:00:00.100772888  4647   0x55a0b65640 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.102193867  4647   0x55a0b65640 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
Pipeline is PREROLLING ...
0:00:00.171533565  4647   0x7f8c19b940 WARN                 default descriptions.c:743:format_info_get_desc: Unexpected MPEG-1 layer in audio/mpeg, mpegversion=(int)1
0:00:00.201158532  4647   0x7f90003d40 WARN                    alsa conf.c:4974:snd_config_expand: alsalib error: Unknown parameters {AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.201244573  4647   0x7f90003d40 WARN                    alsa pcm.c:2495:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.207463395  4647   0x7f90003d40 WARN                    alsa conf.c:4974:snd_config_expand: alsalib error: Unknown parameters {AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.207522602  4647   0x7f90003d40 WARN                    alsa pcm.c:2495:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
mpp[4647]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[4647]: mpp_rt: NOT found ion allocator
mpp[4647]: mpp_rt: found drm allocator
mpp[4647]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[4647]: mpp: unable to create enc h265 for soc rk3399 unsupported
mpp[4647]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[4647]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[4647]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[4647]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[4647]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
0:00:00.502141134  4647   0x7f54055680 WARN               audiosink gstaudiosink.c:218:audioringbuffer_thread_func:<alsasink1> failed to set thread priority
Redistribute latency...
mpp[4647]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[4647]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
mpp[4647]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[4647]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
mpp[4647]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[4647]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
0:00:00.608277857  4647   0x7f540032d0 WARN            kmsallocator gstkmsallocator.c:611:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstAudioSinkClock
0:00:00.805608031  4647   0x7f84003ca0 WARN                  mppdec gstmppdec.c:848:gst_mpp_dec_loop:<mppvideodec0> can't process this frame
mpp[4647]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[4647]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
0:00:00.823612341  4647   0x7f84003ca0 WARN                  mppdec gstmppdec.c:845:gst_mpp_dec_loop:<mppvideodec0> no matched frame
0:00:03.477318845  4647   0x7f900036d0 ERROR                 mppdec gstmppdec.c:941:gst_mpp_dec_handle_frame:<mppvideodec0> failed to send packet
0:00:03.477437551  4647   0x7f900036d0 WARN                  mppdec gstmppdec.c:945:gst_mpp_dec_handle_frame:<mppvideodec0> can't handle this frame
0:00:03.477820795  4647   0x7f8c19b940 WARN             mpegpsdemux gstmpegdemux.c:2945:gst_ps_demux_loop:<mpegpsdemux0> error: Internal data stream error.
0:00:03.477962543  4647   0x7f8c19b940 WARN             mpegpsdemux gstmpegdemux.c:2945:gst_ps_demux_loop:<mpegpsdemux0> error: streaming stopped, reason error (-5)
0:00:03.478373787  4647   0x7f8c19b940 WARN              multiqueue gstmultiqueue.c:2357:gst_multi_queue_sink_event:<multiqueue0> error: Internal data stream error.
0:00:03.478412286  4647   0x7f8c19b940 WARN              multiqueue gstmultiqueue.c:2357:gst_multi_queue_sink_event:<multiqueue0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMpegPSDemux:mpegpsdemux0: Internal data stream error.
Additional debug info:
gstmpegdemux.c(2945): gst_ps_demux_loop (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMpegPSDemux:mpegpsdemux0:
streaming stopped, reason error (-5)
Execution ended after 0:00:02.848726129
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
mpp[4647]: mpp_dec: reset at info change status
mpp[4647]: mpp_dec: task no send to hal que must clr current frame hal status
Setting pipeline to NULL ...
Freeing pipeline ...
gst-launch-1.0[4647]: mpp_mem_pool: put_pool found 1 used buffer size 192
```

LianYangCn, 2022-02-16:
latest MMP + release_20171218 gstreamer-rockchip

All stuff goes well except 卡卡卡 (not smooth)

```log
0:00:17.493652449  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 407 pts=0:00:17.732355555 dts=0:00:17.732355555
0:00:17.526992029  5410   0x7f5c0028a0 TRACE            mppvideodec gstmppvideodec.c:471:gst_mpp_video_dec_loop:<mppvideodec0> finish buffer pts=0:00:17.607233333 dts=0:00:17.607233333
0:00:17.560288942  5410   0x7f5c0028a0 TRACE            mppvideodec gstmppvideodec.c:471:gst_mpp_video_dec_loop:<mppvideodec0> finish buffer pts=0:00:17.648933333 dts=0:00:17.648933333
0:00:17.610283228  5410   0x7f5c0028a0 TRACE            mppvideodec gstmppvideodec.c:471:gst_mpp_video_dec_loop:<mppvideodec0> finish buffer pts=0:00:17.815766666 dts=0:00:17.690644444
0:00:17.643637808  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 408 pts=0:00:17.774066666 dts=0:00:17.774066666
0:00:17.644264882  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 409 pts=0:00:17.940900000 dts=0:00:17.815766666
0:00:17.644879706  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 410 pts=0:00:17.857477777 dts=0:00:17.857477777
0:00:17.645410823  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 411 pts=99:99:99.999999999 dts=0:00:17.899186110
0:00:17.645790858  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 412 pts=0:00:18.066022222 dts=0:00:17.940900000
0:00:17.646071146  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 413 pts=0:00:17.982600000 dts=0:00:17.982600000
0:00:17.646583888  5410   0x7f5c0028a0 TRACE            mppvideodec gstmppvideodec.c:471:gst_mpp_video_dec_loop:<mppvideodec0> finish buffer pts=0:00:17.732355555 dts=0:00:17.732355555
0:00:17.694322750  5410   0x7f5c0028a0 TRACE            mppvideodec gstmppvideodec.c:471:gst_mpp_video_dec_loop:<mppvideodec0> finish buffer pts=0:00:17.774066666 dts=0:00:17.774066666
0:00:17.726939632  5410   0x7f5c0028a0 TRACE            mppvideodec gstmppvideodec.c:471:gst_mpp_video_dec_loop:<mppvideodec0> finish buffer pts=0:00:17.940900000 dts=0:00:17.815766666
0:00:17.777268288  5410   0x7f5c0028a0 TRACE            mppvideodec gstmppvideodec.c:471:gst_mpp_video_dec_loop:<mppvideodec0> finish buffer pts=0:00:17.857477777 dts=0:00:17.857477777
0:00:17.777694698  5410   0x7f5c0028a0 WARN            videodecoder gstvideodecoder.c:2780:gst_video_decoder_prepare_finish_frame:<mppvideodec0> decreasing timestamp (0:00:17.857477777 < 0:00:17.940900000)
0:00:17.810312997  5410   0x7f5c0028a0 TRACE            mppvideodec gstmppvideodec.c:471:gst_mpp_video_dec_loop:<mppvideodec0> finish buffer pts=99:99:99.999999999 dts=0:00:17.899186110
0:00:17.827431488  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 414 pts=0:00:18.024311111 dts=0:00:18.024311111
0:00:17.827883856  5410   0x7f680036d0 DEBUG            mppvideodec gstmppvideodec.c:509:gst_mpp_video_dec_handle_frame:<mppvideodec0> Handling frame 415 pts=0:00:18.191144444 dts=0:00:18.066022222
```

see that?

DTS: 1 2 3 4 5 6 
PTS: 3 1 2 6 4 5

It's drop 1 frame every 3 frames,  
25 FPS * (1-1/3) = 16 FPS

JeffyCN, 2022-02-16:
that log doesn't match the provided mpg.zip. could not reproduce this issue.

LianYangCn, 2022-02-16:
> that log doesn't match the provided mpg.zip. could not reproduce this issue.

you mean this?

```
pi@lanxus-rk3399 ~ $ GST_DEBUG=3,mppvideodec:10  gst-launch-1.0 playbin uri=file:///home/pi/mpg.mpg video-sink=rkximagesink  
Setting pipeline to PAUSED ...
0:00:00.102593723 16618   0x55b3b68640 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.104199873 16618   0x55b3b68640 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
Pipeline is PREROLLING ...
0:00:00.178680954 16618   0x7f9c19b940 WARN                 default descriptions.c:743:format_info_get_desc: Unexpected MPEG-1 layer in audio/mpeg, mpegversion=(int)1
0:00:00.203283600 16618   0x7f90004940 WARN                    alsa conf.c:4974:snd_config_expand: alsalib error: Unknown parameters {AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.203336973 16618   0x7f90004940 WARN                    alsa pcm.c:2495:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.209627119 16618   0x7f90004940 WARN                    alsa conf.c:4974:snd_config_expand: alsalib error: Unknown parameters {AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.209681075 16618   0x7f90004940 WARN                    alsa pcm.c:2495:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
mpp[16618]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[16618]: mpp_rt: NOT found ion allocator
mpp[16618]: mpp_rt: found drm allocator
mpp[16618]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[16618]: mpp: unable to create enc h265 for soc rk3399 unsupported
mpp[16618]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[16618]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[16618]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[16618]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[16618]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
mpp[16618]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[16618]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
mpp[16618]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[16618]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
Redistribute latency...
0:00:00.617277788 16618   0x7f60055280 WARN               audiosink gstaudiosink.c:218:audioringbuffer_thread_func:<alsasink1> failed to set thread priority
0:00:03.054939874 16618   0x7f900042d0 ERROR                 mppdec gstmppdec.c:941:gst_mpp_dec_handle_frame:<mppvideodec0> failed to send packet
0:00:03.055106702 16618   0x7f900042d0 WARN                  mppdec gstmppdec.c:945:gst_mpp_dec_handle_frame:<mppvideodec0> can't handle this frame
0:00:03.055988670 16618   0x7f9c19b940 WARN             mpegpsdemux gstmpegdemux.c:2945:gst_ps_demux_loop:<mpegpsdemux0> error: Internal data stream error.
0:00:03.056185829 16618   0x7f9c19b940 WARN             mpegpsdemux gstmpegdemux.c:2945:gst_ps_demux_loop:<mpegpsdemux0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMpegPSDemux:mpegpsdemux0: Internal data stream error.
Additional debug info:
gstmpegdemux.c(2945): gst_ps_demux_loop (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMpegPSDemux:mpegpsdemux0:
streaming stopped, reason error (-5)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
mpp[16618]: mpp_dec: reset at info change status
mpp[16618]: mpp_dec: task no send to hal que must clr current frame hal status
Freeing pipeline ...
gst-launch-1.0[16618]: mpp_mem_pool: put_pool found 1 used buffer size 192

```

JeffyCN, 2022-02-16:
mpp version is 2021-11-03, which is surely not newest.

JeffyCN, 2022-02-16:
so:
gst mpp 2017 is a out-dated version, there are lots of fixes later(including pts issues)
new gst mpp must work with new mpp and new kernel(might have some unknown dependencies)

LianYangCn, 2022-02-16:
> so: gst mpp 2017 is a out-dated version, there are lots of fixes later(including pts issues) new gst mpp must work with new mpp and new kernel(might have some unknown dependencies)

librockchip_mpp.so.1 and librockchip_mpp.so.2 link to librockchip_mpp.so.0.backup which is backup

my bad!

this is the log
```
pi@lanxus-rk3399 / $ GST_DEBUG=3,mppvideodec:10  gst-launch-1.0 playbin uri=file:///home/pi/mpg.mpg video-sink=rkximagesink
Setting pipeline to PAUSED ...
0:00:00.106415216  2087   0x55bfeb5640 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.107962507  2087   0x55bfeb5640 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
Pipeline is PREROLLING ...
0:00:00.176169027  2087   0x7f7c19b940 WARN                 default descriptions.c:743:format_info_get_desc: Unexpected MPEG-1 layer in audio/mpeg, mpegversion=(int)1
0:00:00.199425937  2087   0x7f80004940 WARN                    alsa conf.c:4974:snd_config_expand: alsalib error: Unknown parameters {AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.199476395  2087   0x7f80004940 WARN                    alsa pcm.c:2495:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
mpp[2087]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[2087]: mpp_rt: NOT found ion allocator
mpp[2087]: mpp_rt: found drm allocator
0:00:00.215417431  2087   0x7f80004940 WARN                    alsa conf.c:4974:snd_config_expand: alsalib error: Unknown parameters {AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.215523015  2087   0x7f80004940 WARN                    alsa pcm.c:2495:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
mpp[2087]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[2087]: mpp: unable to create enc h265 for soc rk3399 unsupported
mpp[2087]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[2087]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[2087]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[2087]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[2087]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
mpp[2087]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[2087]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_start send cmd failed 14
Redistribute latency...
0:00:01.333067203  2087   0x7f40004d40 WARN               audiosink gstaudiosink.c:218:audioringbuffer_thread_func:<alsasink1> failed to set thread priority
mpp[2087]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[2087]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_wait poll cmd failed 110
0:00:03.782341103  2087   0x7f800042d0 ERROR                 mppdec gstmppdec.c:941:gst_mpp_dec_handle_frame:<mppvideodec0> failed to send packet
0:00:03.782586977  2087   0x7f800042d0 WARN                  mppdec gstmppdec.c:945:gst_mpp_dec_handle_frame:<mppvideodec0> can't handle this frame
0:00:03.782943936  2087   0x7f7c19b940 WARN             mpegpsdemux gstmpegdemux.c:2945:gst_ps_demux_loop:<mpegpsdemux0> error: Internal data stream error.
0:00:03.783008686  2087   0x7f7c19b940 WARN             mpegpsdemux gstmpegdemux.c:2945:gst_ps_demux_loop:<mpegpsdemux0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMpegPSDemux:mpegpsdemux0: Internal data stream error.
Additional debug info:
gstmpegdemux.c(2945): gst_ps_demux_loop (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMpegPSDemux:mpegpsdemux0:
streaming stopped, reason error (-5)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
mpp[2087]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[2087]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_start send cmd failed 14
mpp[2087]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[2087]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_wait poll cmd failed 110
Freeing pipeline ...
```

LianYangCn, 2022-02-16:
dmesg output:

```
[  248.710685] rk_vcodec: vpu_service_ioctl:1889: error: unknown vpu service ioctl cmd 40086c01
[  248.749822] rk_vcodec: reg_init:1253: error: translate reg address failed, dumping regs
[  248.757879] rk-vcodec ff650000.vpu_service: reg[00]: 00000000
[  248.763658] rk-vcodec ff650000.vpu_service: reg[01]: 00000000
[  248.769425] rk-vcodec ff650000.vpu_service: reg[02]: 00000000
[  248.775202] rk-vcodec ff650000.vpu_service: reg[03]: 00000000
[  248.780990] rk-vcodec ff650000.vpu_service: reg[04]: 00000000
[  248.786758] rk-vcodec ff650000.vpu_service: reg[05]: 00000000
[  248.792572] rk-vcodec ff650000.vpu_service: reg[06]: 00000000
[  248.798348] rk-vcodec ff650000.vpu_service: reg[07]: 00000000
[  248.804128] rk-vcodec ff650000.vpu_service: reg[08]: 00000000
[  248.809943] rk-vcodec ff650000.vpu_service: reg[09]: 00000000
[  248.815722] rk-vcodec ff650000.vpu_service: reg[10]: 00000000
[  248.821510] rk-vcodec ff650000.vpu_service: reg[11]: 00000000
[  248.827298] rk-vcodec ff650000.vpu_service: reg[12]: 00000000
[  248.833068] rk-vcodec ff650000.vpu_service: reg[13]: 00000000
[  248.838832] rk-vcodec ff650000.vpu_service: reg[14]: 00000000
[  248.844595] rk-vcodec ff650000.vpu_service: reg[15]: 00000000
[  248.850385] rk-vcodec ff650000.vpu_service: reg[16]: 00000000
[  248.856156] rk-vcodec ff650000.vpu_service: reg[17]: 00000000
[  248.861920] rk-vcodec ff650000.vpu_service: reg[18]: 00000000
```

kernal out of date?

JeffyCN, 2022-02-16:
guess so, looks like the kernel driver cannot decode the address from userspace

JeffyCN, 2022-02-16:
tested with newest versions (gst mpp needs https://github.com/JeffyCN/rockchip_mirrors/commit/b30414170c2c5dd7f401819a003b9944c2210497), that mpg plays well on my evb board.

LianYangCn, 2022-02-16:
> guess so, looks like the kernel driver cannot decode the address from userspace

sure?
uname -> 4.4.179

JeffyCN, 2022-02-16:
for the old mpp + old kernel + newest gst mpp issue:

> > try the newest version in: https://github.com/JeffyCN/rockchip_mirrors/tree/gstreamer-rockchip
> > and provide the video file if still has issues.
> 
> Exceptions:
> 
> [image]

try to pick this mpp fix:
https://github.com/JeffyCN/rockchip_mirrors/commit/0d39677317df0aa31aab0e80c39faf79a38bb70a

JeffyCN, 2022-02-16:
> > guess so, looks like the kernel driver cannot decode the address from userspace
> 
> sure? uname -> 4.4.179

the mpp added lots of new features recently(for new boards and new kernels), so it's possible that 2021-11-03 mpp works with your old kernel but 2022-01-26 mpp not.

the mpp guys could only garantee the newest mpp works with newest kernel(which would be tested when releasing newest SDK):
https://github.com/JeffyCN/rockchip_mirrors/tree/kernel

LianYangCn, 2022-02-16:
linux kernel: 4.4.179

1. build and install:

```bash
git clone -b mpp --single-branch  https://github.com/JeffyCN/rockchip_mirrors.git mpp_mirrors
git clone -b gstreamer-rockchip --single-branch  https://github.com/JeffyCN/rockchip_mirrors.git gstreamer-rockchip
```

2. run:
```
GST_DEBUG=3,mppvideodec:10  gst-launch-1.0 playbin uri=file:///home/pi/mpg.mpg video-sink=rkximagesink
```
3. got:
```
Setting pipeline to PAUSED ...
0:00:00.106985907 10429   0x557a98a240 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.108541919 10429   0x557a98a240 WARN                 basesrc gstbasesrc.c:3583:gst_base_src_start_complete:<source> pad not activated yet
Pipeline is PREROLLING ...
0:00:00.168694414 10429   0x7f9019b540 WARN                 default descriptions.c:743:format_info_get_desc: Unexpected MPEG-1 layer in audio/mpeg, mpegversion=(int)1
0:00:00.201116013 10429   0x7f94004940 WARN                    alsa conf.c:4974:snd_config_expand: alsalib error: Unknown parameters {AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.201204095 10429   0x7f94004940 WARN                    alsa pcm.c:2495:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
mpp[10429]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[10429]: mpp_rt: NOT found ion allocator
mpp[10429]: mpp_rt: found drm allocator
mpp[10429]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[10429]: mpp: unable to create enc h265 for soc rk3399 unsupported
mpp[10429]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[10429]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
mpp[10429]: mpp_info: mpp version: 8c13cd93 author: Hongjin Li    2022-01-26 [mpi_enc_test]: add slt testcase
0:00:00.264641903 10429   0x7f94004940 WARN                    alsa conf.c:4974:snd_config_expand: alsalib error: Unknown parameters {AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
0:00:00.264708110 10429   0x7f94004940 WARN                    alsa pcm.c:2495:snd_pcm_open_noupdate: alsalib error: Unknown PCM default:{AES0 0x02 AES1 0x82 AES2 0x00 AES3 0x02}
mpp[10429]: mpp_buf_slot: set frame info: w  720 h  480 hor  720 ver  480
mpp[10429]: mpp_dec: setting default w  720 h  480 h_str  720 v_str  480
mpp[10429]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[10429]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_start send cmd failed 14
Redistribute latency...
0:00:01.400434774 10429   0x7f50004540 WARN               audiosink gstaudiosink.c:218:audioringbuffer_thread_func:<alsasink1> failed to set thread priority
mpp[10429]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[10429]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_wait poll cmd failed 110
mpp[10429]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[10429]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_start send cmd failed 14
mpp[10429]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[10429]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_wait poll cmd failed 110
mpp[10429]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[10429]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_start send cmd failed 14
mpp[10429]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[10429]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_wait poll cmd failed 110
0:00:09.807554204 10429   0x7f940042d0 ERROR                 mppdec gstmppdec.c:947:gst_mpp_dec_handle_frame:<mppvideodec0> failed to send packet
0:00:09.807681952 10429   0x7f940042d0 WARN                  mppdec gstmppdec.c:951:gst_mpp_dec_handle_frame:<mppvideodec0> can't handle this frame
0:00:09.808250399 10429   0x7f9019b540 WARN             mpegpsdemux gstmpegdemux.c:2945:gst_ps_demux_loop:<mpegpsdemux0> error: Internal data stream error.
0:00:09.808355981 10429   0x7f9019b540 WARN             mpegpsdemux gstmpegdemux.c:2945:gst_ps_demux_loop:<mpegpsdemux0> error: streaming stopped, reason error (-5)
ERROR: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMpegPSDemux:mpegpsdemux0: Internal data stream error.
Additional debug info:
gstmpegdemux.c(2945): gst_ps_demux_loop (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstMpegPSDemux:mpegpsdemux0:
streaming stopped, reason error (-5)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
mpp[10429]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address
mpp[10429]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_start send cmd failed 14
mpp[10429]: vcodec_service: vcodec_service_cmd_poll ioctl VPU_IOC_GET_REG failed ret -1 errno 110 Connection timed out
mpp[10429]: hal_m2vd_vdpu2: hal_m2vd_vdpu2_wait poll cmd failed 110
Freeing pipeline ...
```

JeffyCN, 2022-02-16:
as mentioned above, 2022-01-26 mpp could not work with your old 4.4.179 kernel.

closed 2022-02-16 by JeffyCN

LianYangCn, 2022-02-16:
> the mpp added lots of new features recently(for new boards and new kernels), so it's possible that 2021-11-03 mpp works with your old kernel but 2022-01-26 mpp not.
>
> the mpp guys could only garantee [...]

bad guys

I can find out the kernel source code to merge with https://github.com/JeffyCN/rockchip_mirrors/tree/kernel
the modification I know:
1. enable EDP display output --> easy
2. add a splash image --> easy
3. 6 LEDs (connect to GPIOs) show current state --> hell

JeffyCN, 2022-02-16:
so you need newest gstreamer-rockchip plugins.

and for mpp+kernel's compatible issue, there're 2 ways to solve it:
1/ update your kernel, and use the newest mpp

or
2/ use your current kernel + 2021-11-03 mpp (which should work according to your previous reply), then pick this fix:
https://github.com/JeffyCN/rockchip_mirrors/commit/0d39677317df0aa31aab0e80c39faf79a38bb70a

LianYangCn, 2022-02-16:
> so you need newest gstreamer-rockchip plugins for the PTS issue.
> 
> and for mpp+kernel's compatible issue, there're 2 ways to solve it: 1/ update your kernel, and use the newest mpp
> 
> or 2/ use your current kernel + current mpp (2021-11-03, which should work according to your previous reply), then pick this fix: [JeffyCN/rockchip_mirrors@0d39677](https://github.com/JeffyCN/rockchip_mirrors/commit/0d39677317df0aa31aab0e80c39faf79a38bb70a)

2. use current kernel(4.4.179) + current mpp(2021-11-03) and pitch fix 0d39677
  + https://github.com/JeffyCN/rockchip_mirrors/tree/gstreamer-rockchip not work
  + https://github.com/rockchip-linux/gstreamer-rockchip/tree/release_20171218 work, but there is PTS issue.